dresden-elektronik/deconz-rest-plugin #7526: Add support for Xiaomi/Azarton MHO-C401 Temperature and Humidity Sensor

andreas5232, 2024-01-10:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Xiaomi MMC MHO-C401

### Manufacturer

Xiaomi

### Model identifier

MHO-C401

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]
[image]
[image]

### Basic

[image]

### Further relevant clusters

### Any other cluster of relevance/interest

This is my first device request for deconz, so any help or feedback is appreciated. :-)

I've already built a POC for a DDF file (based on #7327 which is a similar Xiaomi sensor):
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "MiaMiaoCe",
  "modelid": "MHO-C401N",
  "vendor": "Xiaomi",
  "product": "MHO-C401N",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/mho-c401n.json",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true,
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405"
    }
  ]
}
```

SwoopX, 2024-01-10:
Hi,

thanks for raising the request. The suggested DDF looks already pretty good, so you might want to issue a PR for it and consider the binding settings as with the referenced device here (at the end) https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7373/files

andreas5232, 2024-01-13:
> thanks for raising the request. The suggested DDF looks already pretty good, so you might want to issue a PR for it and consider the binding settings as with the referenced device here (at the end) https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7373/files

Thank your for your kind feedback!  When using the bindings from the other pull request, the sensor updates now seem to be more battery friendly. I changed those in my PR as suggested. Now the sensor updates are fetched once in a while every few minutes, which seems to be more useful to me.

SwoopX, 2024-01-13:
Sounds good, thanks! I'll check it.

closed 2024-01-17 by manup