dresden-elektronik/deconz-rest-plugin #7531: Pairing Senic Firends of Hue Switch on macOS results in a crash

LoPablo, 2024-01-15:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Traing to paire the Senic Switch on macOS results in deconz crashing right the moment the button of the switch is let go. 

The last message that is display in the console is ```/Applications/deCONZ.app/Contents/MacOS/deCONZ is loading libcrypto in an unsafe way```

### Steps to reproduce the behavior

1. Start deconz
2. Start pairing process for Friends of Hue switch
3. Push Button on Switch
4. Hold for 10 seconds
5. let go
6. crash

### Expected behavior

Switch is paired, deconz does not crash

### Environment

- Host system:  macOS
- deCONZ version: 2.25.1
- Device: ConBee II
- Do you use an USB extension cable: yes

LoPablo, 2024-01-20:
Here is the stacktrace from the MacOs Error-Reporter if this helps

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               deCONZ [14599]
Path:                  /Applications/deCONZ.app/Contents/MacOS/deCONZ
Identifier:            de.phoscon.deconz
Version:               2.22.0 (???)
Code Type:             X86-64 (Native)
Parent Process:        zsh [14513]
Responsible:           Terminal [23980]
User ID:               501

Date/Time:             2024-01-20 18:43:09.4139 +0100
OS Version:            macOS 13.4.1 (22F770820d)
Report Version:        12
Anonymous UUID:        7FC14C94-2776-C0FC-D432-5CEACE4527F7

Time Awake Since Boot: 680000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called
Invalid dylib load. Clients should not load the unversioned libcrypto dylib as it does not have a stable ABI.

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff8021851f2 __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff8021bcee6 pthread_kill + 263
2   libsystem_c.dylib             	    0x7ff8020e3b45 abort + 123
3   libcrypto.dylib               	    0x7ffc06be3aee __report_load.cold.1 + 36
4   libcrypto.dylib               	    0x7ffc06be3884 __report_load + 569
5   dyld                          	    0x7ff801e7c41b invocation function for block in dyld4::Loader::findAndRunAllInitializers(dyld4::RuntimeState&) const::$_0::operator()() const + 175
6   dyld                          	    0x7ff801ebab6a invocation function for block in dyld3::MachOAnalyzer::forEachInitializer(Diagnostics&, dyld3::MachOAnalyzer::VMAddrConverter const&, void (unsigned int) block_pointer, void const*) const + 242
7   dyld                          	    0x7ff801eaef22 invocation function for block in dyld3::MachOFile::forEachSection(void (dyld3::MachOFile::SectionInfo const&, bool, bool&) block_pointer) const + 577
8   dyld                          	    0x7ff801e5f0af dyld3::MachOFile::forEachLoadCommand(Diagnostics&, void (load_command const*, bool&) block_pointer) const + 245
9   dyld                          	    0x7ff801eae0bf dyld3::MachOFile::forEachSection(void (dyld3::MachOFile::SectionInfo const&, bool, bool&) block_pointer) const + 175
10  dyld                          	    0x7ff801eba72a dyld3::MachOAnalyzer::forEachInitializer(Diagnostics&, dyld3::MachOAnalyzer::VMAddrConverter const&, void (unsigned int) block_pointer, void const*) const + 470
11  dyld                          	    0x7ff801e7968c dyld4::Loader::findAndRunAllInitializers(dyld4::RuntimeState&) const + 220
12  dyld                          	    0x7ff801e821f4 dyld4::PrebuiltLoader::runInitializers(dyld4::RuntimeState&) const + 30
13  dyld                          	    0x7ff801e7987a dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&) const + 178
14  dyld                          	    0x7ff801e7c48f dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const::$_1::operator()() const + 97
15  dyld                          	    0x7ff801e79911 dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const + 93
16  dyld                          	    0x7ff801e963f6 dyld4::APIs::dlopen_from(char const*, int, void*) + 944
17  QtCore                        	       0x10ee67ef0 0x10ec76000 + 2039536
18  QtCore                        	       0x10ee632cb 0x10ec76000 + 2020043
19  libde_rest_plugin.dylib       	       0x126391932 GP_DecryptSecurityKey(unsigned int, std::__1::array<unsigned char, 16ul> const&) + 290
20  libde_rest_plugin.dylib       	       0x126342842 DeRestPluginPrivate::gpDataIndication(deCONZ::GpDataIndication const&) + 1826
21  QtCore                        	       0x10eea20d5 0x10ec76000 + 2277589
22  libdeCONZ.1.2.0.dylib         	       0x10cc9c2e4 deCONZ::GreenPowerController::gpDataIndication(deCONZ::GpDataIndication const&) + 52
23  libdeCONZ.1.2.0.dylib         	       0x10cca5987 deCONZ::GreenPowerController::processIncomingData(QByteArray const&) + 231
24  deCONZ                        	       0x10cade0d7 zmMaster::processPacked(zm_command const*) + 2263
25  deCONZ                        	       0x10cae2bc2 0x10ca5a000 + 560066
26  deCONZ                        	       0x10ca69bda protocol_receive + 282
27  deCONZ                        	       0x10cae26c8 SerialCom::readyRead() + 648
28  deCONZ                        	       0x10cae22ee 0x10ca5a000 + 557806
29  QtCore                        	       0x10ee99b9f QObject::event(QEvent*) + 943
30  QtWidgets                     	       0x10d4619ea QApplicationPrivate::notify_helper(QObject*, QEvent*) + 266
31  QtWidgets                     	       0x10d462e11 QApplication::notify(QObject*, QEvent*) + 497
32  QtCore                        	       0x10ee6ea34 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
33  QtCore                        	       0x10ee6fd79 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 809
34  libqcocoa.dylib               	       0x10da4b259 0x10da11000 + 238169
[... 62 lines cut ...]
6   libsystem_pthread.dylib       	    0x7ff8021b8bd3 thread_start + 15

Thread 6:: QNetworkAccessManager thread
[... cut ...]
  rax: 0x0000000000000000  rbx: 0x00007ff845899640  rcx: 0x00007ff7b34a0fb8  rdx: 0x0000000000000000
  rdi: 0x0000000000000103  rsi: 0x0000000000000006  rbp: 0x00007ff7b34a0fe0  rsp: 0x00007ff7b34a0fb8
   r8: 0x00007ff8458a0d18   r9: 0x0000000000000000  r10: 0x0000000000000000  r11: 0x0000000000000246
  r12: 0x0000000000000103  r13: 0x00007ff7b34a10d0  r14: 0x0000000000000006  r15: 0x0000000000000016
  rip: 0x00007ff8021851f2  rfl: 0x0000000000000246  cr2: 0x00007ff843b245b8
  
Logical CPU:     0
Error Code:      0x02000
```

LoPablo, 2024-01-20:
So this seems to be a problem in [green_power.ccp](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/green_power.cpp) which tried to load libcrypt. I am currently trying to specify the path to libcrypt and debug the issue but I am currently unable to get deconz to recognise my compiled plugin.dylib.

LoPablo, 2024-03-05:
This has not been fixed yet. Please reopen.

manup, 2024-03-24:
OpenSSL on macOS is now fixed via [v2.26.2-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.26.2-beta), closing for now.

closed 2024-03-24 by manup

LoPablo, 2024-03-25:
Can confirm that pairing a switch now works