dresden-elektronik/deconz-rest-plugin #7533: Support for new Namron 8 button switch

joernnilsson, 2024-01-16:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Zigbee Channel Switch

### Manufacturer

Namron AS

### Model identifier

4512772

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]
[image]

### Basic

[image]

There are 4 basic clusters, they repeat the same information

### Further relevant clusters

Power
[image]

Diagnostic
[image]

Manufacturor page and user manual:
https://www.namron.com/products/namron-zigbee-kanalbryter-hvit/
https://www.elektroimportoren.no/docs/lib/4512772-Brukerveiledning-5.pdf

This appears to be a new version of Namron's 8 button switch. The old one appears in Phoscon as expected.

Mimiix, 2024-02-10:
@Smanar Perhaps you want to check?

Smanar, 2024-02-11:
This one don't have dimmer like the 4512721 ?
Because both device relay look same (zigbee side speaking)
Can make a try just with adding the model in this DDF `devices\sunricher\8button_switch.json`

And in buttonmap file here

```
        "sunricherMap": {
            "vendor": "Sunricher",
            "doc": "Wireless switches from Sunricher, Namron, SLC and EcoDim",
            "modelids": ["ED-10010", "ED-10011", "ED-10012", "ED-10013", "ED-10014", "ED-10015", "ZG2833K2_EU07", "ZG2833K8_EU05", "ZG2833K4_EU06", "ZG2835", "4512701", "4512702", "4512703", "4512705", "4512719", "4512721", "4512728", "4512729", "S57003", "ROB_200-008", "ROB_200-009-0", "ROB_200-008-0", "ROB_200-007-0"],
            "buttons": [
```

joernnilsson, 2024-02-16:
I can confirm that the two changes works perfectly :tada: 

Thank you very much @Smanar and @Mimiix.

I've created a PR with the changes: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7615

closed 2024-03-16 by github-actions[bot]