dresden-elektronik/deconz-rest-plugin #7539: Schneider Electric / Merten Raumtemperaturregler-Modul, polarweiss, System M, ZB

sdriesche, 2024-01-20:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Connected Raumtemperaturregler- Modul, polarweiss, System M, ZB

### Manufacturer

Schneider Electric

### Model identifier

MEG5779

### Device type to add

Thermostat

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Thermostat User Interface Configuration Cluster
[image]

### 02 Home Automation Temperature Sensor - Basic
[image]
[image]

### 02 Home Automation Temperature Sensor  - Temperature Measurement
[image]

### 04 Home Automation Occupancy sensor - Basic
[image]
[image]

### 04 Home Automation Occupancy sensor - Occupancy sensing
[image]

### 05 Home Automation Consumption Awareness Device - Basic
[image]
[image]
 
### 05 Home Automation Consumption Awareness Device - 0702 Simple Metering
[image]
[image]

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

closed 2024-02-19 by github-actions[bot]

sdriesche, 2024-02-19:
Hi, is there a quick way to get the basics (thermometer read-out, temperature set, and valve setting) of the thermostat working?

Smanar, 2024-02-22:
And I was wrong, you can keep the TYPE_CONSUMPTION_SENSOR.
But Now I m looking better the DDF, but was same device ?
It use the TYPE_CONSUMPTION_SENSOR on same endpoint than your, the 0x05 ....