dresden-elektronik/deconz-rest-plugin #7541: Philips Hue Signe floor lamp

BEskandari, 2024-01-21:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Philips Hue Signe

### Manufacturer

Philips

### Model identifier

4080248P9

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]
[image]

### Further relevant clusters

### Identify
[image]

### Groups
[image]
[image]

### Scenes
[image]
[image]
[image]
[image]

### On/off
[image]
[image]
[image]

### Level Control
[image]
[image]
[image]

### Touchlink Commissioning
[image]

### Hue Effects
[image]

### Color Control
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Hue Entertainment
[image]

BEskandari, 2024-01-29:
@ebaauw, by any chance do you think that with an existing Philips Hue DDF we can control this device?

ebaauw, 2024-01-29:
Yeah, probably just another ZB3 gamut C light.  You might want to check the _Image Type_.

BEskandari, 2024-01-30:
Thanks for the hint. I added the model identifier in light_zb3_C.json and it is working fine

PR #7562 (2024-02-05): DDF for Philips Signe floor lamp (4080248P9)
close #7541
closed 2024-02-05 by manup