dresden-elektronik/deconz-rest-plugin #7542: Add Tuya Smart Zigbee Thermometer/Humidy reader

KRGT-NL, 2024-01-21:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya Smart Zigbee Thermometer

### Manufacturer

_TZE200_zl1kmjqx

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
Ordered: https://nl.aliexpress.com/item/1005002836127648.html

KRGT-NL, 2024-01-21:
As you can see i do not get enough information from this device to fill in all details...

Mimiix, 2024-01-21:
Can you share the list of clusters ?

KRGT-NL, 2024-01-21:
> Can you share the list of clusters ?
Do you mean this one?
[image]

Mimiix, 2024-01-21:
> > Can you share the list of clusters ?
> > Do you mean this one?
> > ![image](https://private-user-images.githubusercontent.com/86347728/298393730-68b53aa9-d994-4bb8-b731-a5c092dae3a3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU4NTc3ODYsIm5iZiI6MTcwNTg1NzQ4NiwicGF0aCI6Ii84NjM0NzcyOC8yOTgzOTM3MzAtNjhiNTNhYTktZDk5NC00YmI4LWI3MzEtYTVjMDkyZGFlM2EzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIxVDE3MTgwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI1NmFiOTZhZDkyNDYyOTQ4YzRhZWY0ODI5NDhiMjEzMWY5MWQ4N2ZiZmJiZTZiM2MxYjc4OGUxMWM4ZmJmOWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5K0Mcubj8-YFao5jmwoGPoJsAdLCGy9YcwvZTT3Q8NE)
> 
> As you can see it is not goodly recognized, as it;s just a sensor in all means, like the Ikea it misses a lot of clusters. ![image](https://private-user-images.githubusercontent.com/86347728/298393982-f156ec28-9827-45de-9652-5c94e60eec4c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU4NTc3ODYsIm5iZiI6MTcwNTg1NzQ4NiwicGF0aCI6Ii84NjM0NzcyOC8yOTgzOTM5ODItZjE1NmVjMjgtOTgyNy00NWRlLTk2NTItNWM5NGU2MGVlYzRjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIxVDE3MTgwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ4ZjFhYTgxNjcwNTliMmVhMGJkYjU4NmI3ZmRkODQyYTRiN2FkNDM5OTE2MjcxNzg3YzdjZDg4MWI0YTdiNjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.a68hST7CNUp6i45z0rV00YAe4riArF4ZLtXrfnw8KTk)

Thats often how Tuya works :) no problem there.

They just don't stick to the standard...

KRGT-NL, 2024-02-12:
No update?

Mimiix, 2024-02-12:
@Smanar Can you check in / help out :)?

KRGT-NL, 2024-02-25:
Let me check how I do this!

Smanar, 2024-02-25:
Depend of your OS, easier on real OS.

`All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.`

If you are on the offical docker, you have a persistent folder.

If you are using HA, you have one too, but harder to use.

KRGT-NL, 2024-02-25:
You can guess which one I have haha

I will check with Portainer the exact mapping in HassOS and will put the file on Hass with the Samba Share addon :-)

Smanar, 2024-02-25:
Lol.
you have the short and long procedure here https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

KRGT-NL, 2024-03-09:
[image]

Looks like that it works. :)

closed 2024-03-09 by KRGT-NL

Smanar, 2024-03-09:
Thx for return.
PR done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7658