dresden-elektronik/deconz-rest-plugin #7544: ZSHARK reports "device not open" on connect

danube, 2024-01-21:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I've downloaded and installed ZSHARK on Raspberry OS. When I launch the GUI via shell, the applicaiton shows up with only one device in the dropdown: ConBee. When I choose "Connect", nothing happens and the application freezes. When I launch zshark from the shell, after clicking "Connect", it reveals the message: `IODevice::write (QSerialPort): device not open`. I've attached a ConBee III on the Raspi's USB port and am running a network with 38 nodes. Any help in getting ZSHARK to work?

### Steps to reproduce the behavior

1. Having network with devices on Raspberry Pi OS with ConBee III running.
2. Install ZSHARK as instructed [here](https://phoscon.de/en/conbee/software#zshark).
3. Run `zshark` from the command line.
4. In the GUI, choose "Connect".

### Expected behavior

Application connects to device

### Environment

- Host system: Raspberry Pi: Linux rpi4 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
- ConBee III Firmware: 264F0900
- deCONZ version: 2.24.3
- USB extension cable 1m
- No other USB or serial devices connected.

Mimiix, 2024-01-21:
Not a bug, as zshark is only supported for conbee and Conbee 2.

The instructions you link are for conbee 1.

Closing.

danube, 2024-01-21:
That's ahm... not obvious. Thanks anyway. Can you provice any ressource of information how to get ZSHARK running with ConBee 3?

Mimiix, 2024-01-21:
> That's ahm... not obvious. Thanks anyway. Can you provice any ressource of information how to get ZSHARK running with ConBee 3?

You can not as there's no firmware for it at this moment.

No idea when that would come.

danube, 2024-01-22:
Too bad, thanks anyway.

closed 2024-01-22 by danube