dresden-elektronik/deconz-rest-plugin #7547: Raspbee II under Windows (11)

PatcherMustermann, 2024-01-23:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I would like to use the Raspbee II under Windows (11). However, I am not able to integrate it as a device in deConz. 

I have tried the following parameters:
01 - deconz -d COMx
02 - deconz -d "COMx"
03 - deconz -d \\.\COMx
04 - deconz -d "\\.\COMx"
05 - deconz --dev=COMx
06 - deconz --dev="COMx"
07 - deconz --dev=\\.\COMx
08 - deconz --dev="\\.\COMx"
09 - deconz --device=COMx
10 - deconz --device="COMx"
11 - deconz --device=\\.\COMx
12 - deconz --device="\\.\COMx"

I have run COM-Port monitoring and deConz does not even try to open the COM-Port. Under Zigbee2MQTT (Windows), however, the Raspbee II works perfectly.

With the GCFFlasher_Win_V4_03_00 (Windows) I can establish a connection with the parameter "-d" (GCFFlasher.exe -d COMx). With the list function "-l", however, the Raspbee II is not displayed either.

### Steps to reproduce the behavior

/

### Expected behavior

/

### Environment

- Host system:  PC (Windows 11)
- Device:  RaspBee II

manup, 2024-02-05:
Indeed the `--dev` parameter currently just works on Linux, so right know the best approach is to use the --auto-connect=1 (which is the default) and let deCONZ figure out the COM port. Note on Windows there is no RaspBee, do you mean ConBee?

PatcherMustermann, 2024-02-05:
I mean realy the Raspbee II. I have successfully connect the Raspbee II to deconz on Windows with a trick, but for this I need a Conbee on the USB port. Why does the device parameter only work in Linux and not in Windows? I think for dresden elektronik is this not a big deal to change the source code for the deconz Windows version.

manup, 2024-02-05:
Yeah I think adapting the code to also support `--dev` on Windows doesn't require much. I put it on schedule for v2.26.0-beta. Note however that you might also need to set the `--baudrate=38400` to connect to RaspBee.

PatcherMustermann, 2024-02-06:
That sounds good. Thank you very much!

Would the call of deconz with parameters look like this: \path...\deconz.exe --dev=COMx --baudrate=38400

manup, 2024-02-06:
Yeah I guess the call looks right if it is connected to a USB serial converter.

All versions are also released for Windows like the latest beta v2.25.2. However build instructions needs to be updated for Windows since the stock Qt5 libraries need to be used, unfortunately the msys2 Qt5 libraries generate bogus virus warnings.

PatcherMustermann, 2024-02-13:
I tried the releases v2.26.0-beta from yesterday and my problem is not fixed yet. Is this correct?
[image]

Mimiix, 2024-02-13:
> I tried the releases v2.26.0-beta from yesterday and my problem is not fixed yet. Is this correct?
> 
> ![Screenshot 2024-02-13 103034](https://private-user-images.githubusercontent.com/15119397/304360403-9ef7b0c2-49dd-4381-b774-f5c60a42d536.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc4MTc2NTQsIm5iZiI6MTcwNzgxNzM1NCwicGF0aCI6Ii8xNTExOTM5Ny8zMDQzNjA0MDMtOWVmN2IwYzItNDlkZC00MzgxLWI3NzQtZjVjNjBhNDJkNTM2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjEzVDA5NDIzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc5MmEwYjAzZDVmZGE1MDBjNDZlYjU5YWI3NWMzY2FiYTZlMGE2Y2IwMDVkNGQwMmM1M2U0ZGYzODY0ZWQ5ZmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZnUq7-TV7IoB9vGeLsTQsTY2HkGW2qckpmL5rFDBJ-Y)

Yes. Otherwise this issue would have been closed.

PatcherMustermann, 2024-02-13:
Thanks for this information. Then I'll have to be patient a little longer.

PatcherMustermann, 2024-03-01:
I tested version 2.26.1 beta today and the parameter —dev=COMx worked fine. One thing that could be improved is to change the displayed name ConBee to Rasbee. Otherwise very good job - thank you!

manup, 2024-03-28:
Close as completed for now, the RaspBee device name needs some more additions in the firmware later on.

closed 2024-03-28 by manup