dresden-elektronik/deconz-rest-plugin #7550: All zigbee devices stop responding

DennisMTB, 2024-01-26:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Since yesterday 25-1, my zigbee devices connected to my Conbee2 Gateway stopped working. In my experience, these fell out one by one. The strange thing is that sometimes (after reboot) both Phoscon and Home Assistant devices are shown as connected (not gray) but they do not respond individualy. I can turn off all the lights in the Phoscon app (All off). Lamps, switches and sensors do not work.

[image]

### Steps to reproduce the behavior

I have allready did the following steps:

1. Rebooted Home Assistant compleet system (multiple times)
2. Load last configuration in de Phoscon Gateway (hidden window)
3. Restore a backup (configuration of the Gateway)
4. Tried new Hardware (HA back-up machine with new Conbee 2 stick)
5. Power off Multiple lights 

### Expected behavior

1. End devices responding in Phoscon app
2. Devices responding in Home Assistant

### Screenshots

[image]
[image]
[image]
[image]
[image]
[image]

### Environment

- Host system: Raspberry P4 (running without any issues for 3 years)
- Running method: Home Assistent deCONZ Add-on (latest version)
- Firmware version: (2.22.02)
- deCONZ version 2.22.2
- Device:  ConBee II 
- Do you use an USB extension cable: YES
- Is there any other USB or serial devices connected to the host system? SSD (extension cable)

### deCONZ Logs

After reboot:

2:14:21:969 wait reconnect 15 seconds
12:14:22:092 aps request id: 5 failed, erase from queue
12:14:22:470 networkState: CC_ReconnectNetwork
12:14:22:471 start reconnect to network
12:14:22:970 wait reconnect 14 seconds
12:14:23:969 wait reconnect 13 seconds
12:14:24:970 wait reconnect 12 seconds
12:14:25:969 wait reconnect 11 seconds
12:14:26:969 wait reconnect 10 seconds
12:14:27:722 failed to reconnect to network try=1
12:14:27:969 wait reconnect 9 seconds
12:14:28:970 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
12:14:28:971 wait reconnect 8 seconds
12:14:29:969 wait reconnect 7 seconds
12:14:30:969 wait reconnect 6 seconds
12:14:31:970 wait reconnect 5 seconds
12:14:32:970 failed to reconnect to network try=2
12:14:32:971 wait reconnect 4 seconds
[... 255 lines cut ...]
12:16:11:774 APS-DATA.request id: 114, addrmode: 0x03, addr: 0x00212effff069d58, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
12:16:11:775 	asdu (length: 2): fb00
12:16:11:807 APS-DATA.confirm id: 114, status: 0x00 SUCCESS
12:16:11:808 APS-DATA.confirm request id: 114 -> confirmed, timeout 3405806
12:16:11:808 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 221, rssi: 30
12:16:11:809 	asdu: fb00000000
12:16:11:809 APS-DATA.indication request id: 114 -> finished
12:16:11:810 APS-DATA.request id: 114 erase from queue
12:16:15:134 APS-DATA.request id: 116, addrmode: 0x03, addr: 0x00212effff069d58, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
12:16:15:135 	asdu (length: 2): fc00
12:16:15:168 APS-DATA.confirm id: 116, status: 0x00 SUCCESS
12:16:15:169 APS-DATA.confirm request id: 116 -> confirmed, timeout 3409166
12:16:15:170 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 221, rssi: 30
12:16:15:171 	asdu: fc00000000
12:16:15:172 APS-DATA.indication request id: 116 -> finished
12:16:15:172 APS-DATA.request id: 116 erase from queue
12:16:18:519 APS-DATA.request id: 120, addrmode: 0x03, addr: 0x00212effff069d58, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
12:16:18:521 	asdu (length: 2): fd00
12:16:18:541 APS-DATA.confirm id: 120, status: 0x00 SUCCESS
12:16:18:541 APS-DATA.confirm request id: 120 -> confirmed, timeout 3412552
12:16:18:586 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 221, rssi: 30
12:16:18:587 	asdu: fd00000000
12:16:18:587 APS-DATA.indication request id: 120 -> finished
12:16:18:588 APS-DATA.request id: 120 erase from queue
12:16:18:969 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0

DennisMTB, 2024-01-26:
This is my Raspberry Hardware config (of the Conbee2)

ttyACM0
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2403671-if00
Subsysteem:
tty
Apparaatpad:
/dev/ttyACM0
ID:
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2403671-if00
Attributen:
DEVLINKS: >-
  /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2403671-if00
  /dev/serial/by-path/platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.3:1.0
DEVNAME: /dev/ttyACM0
DEVPATH: >-
  /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/tty/ttyACM0
ID_BUS: usb
ID_MODEL: ConBee_II
ID_MODEL_ENC: ConBee\x20II
ID_MODEL_ID: '0030'
ID_PATH: platform-fd500000.pcie-pci-0000:01:00.0-usb-0:1.3:1.0
ID_PATH_TAG: platform-fd500000_pcie-pci-0000_01_00_0-usb-0_1_3_1_0
ID_REVISION: '0100'
ID_SERIAL: dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2403671
ID_SERIAL_SHORT: DE2403671
ID_TYPE: generic
ID_USB_DRIVER: cdc_acm
ID_USB_INTERFACES: ':020201:0a0000:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: ConBee_II
ID_USB_MODEL_ENC: ConBee\x20II
ID_USB_MODEL_ID: '0030'
ID_USB_REVISION: '0100'
ID_USB_SERIAL: dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2403671
ID_USB_SERIAL_SHORT: DE2403671
ID_USB_TYPE: generic
ID_USB_VENDOR: dresden_elektronik_ingenieurtechnik_GmbH
ID_USB_VENDOR_ENC: dresden\x20elektronik\x20ingenieurtechnik\x20GmbH
ID_USB_VENDOR_ID: 1cf1
ID_VENDOR: dresden_elektronik_ingenieurtechnik_GmbH
ID_VENDOR_ENC: dresden\x20elektronik\x20ingenieurtechnik\x20GmbH
ID_VENDOR_ID: 1cf1
MAJOR: '166'
MINOR: '0'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '3299900203'

Mimiix, 2024-01-26:
Clearly not a bug. Please use the forums.