dresden-elektronik/deconz-rest-plugin #7551: Philips Hue White Ambiance (LTW001) change state to „on“ automatically

FlatTV, 2024-01-27:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

After switching off the lamp using the Deconz API, the status in the API changes back to “On” after a few seconds. In reality the lamp is still off.

### Steps to reproduce the behavior

1. Switch lamp to „off“ in deconz-API
2. The lamp switch to „off“ and the state in deconz-API switch to „off“.
3. wait and do nothing…
4. The deconz API state switch to „on“ but the lamp is still „off“.

### Expected behavior

just nothing

### Screenshots

[https://forum.phoscon.de/t/ltw001-light-shows-on-after-vew-seconds-in-deconz-api-but-the-real-state-is-off/4378](url)

### Environment

- Host system: (Raspberry Pi)
- Running method: (Raspbian)
- Firmware version: (26720700)
- deCONZ version (not Home assistant Addon version!): (2.25.1)
- Device: (ConBee II)
- Do you use an USB extension cable: (no)
- Is there any other USB or serial devices connected to the host system? If so: Which? (no)

SwoopX, 2024-01-27:
Hm, interesting case. Could you please fetch some debug logs?

#### Option 1:
To get debug logs in a GUI installation, open deCONZ and click `Help`. After that, click `Debug View`. The following debug levels need to be enabled for proper logging: `INFO`, `INFO_L2`, `ERROR`, `ERROR_L2`, `APS`, `APS_L2`.
Please note that in the HA Addon and Docker webVNC, you can't copy paste. You need to use a Native VNC Client. [HA Addon How-To](https://community.home-assistant.io/t/deconz-addon-how-to-enable-vnc-set-port-in-network-configuration/137597), [Docker Environment Vars](https://github.com/deconz-community/deconz-docker#environment-variables).

#### Option 2:
To get debug logs in a headless installation / ssh session, use the following commands:

```
sudo systemctl stop deconz-gui
sudo systemctl stop deconz
/usr/bin/deCONZ -platform minimal --dbg-aps=2 --dbg-info=2 --dbg-error=2 | tee debug.txt
```

When done, exit the process with `CTRL+C`. You can now start the systemctl service again. Your logs will be printed in the console, as well as saved to a file named `debug.txt` in the current directory. Be sure to capture at least a few minutes worth of logs. When sharing these logs, DO NOT attempt to paste them into a channel. Instead use a pastebin service such as [this](https://paste.debian.net/) or [this](https://pastebin.com/).

Afterwards, don't forget to disable logging. This is especially important when running off an SD card, as the debug logging will cause unnessecary writes, and might shorten the life of your media.

FlatTV, 2024-02-01:
Log: Lamp “WZ_Schranklicht” has the status “On”, then I switch it to “Off” via API.
[https://pastebin.com/pfuwTs37](url)

SwoopX, 2024-02-02:
Thanks!

@ebaauw This might be one for you. I assume line 2767 may give the answer, at least the stats seems to change only through cluster 0xfc03.

ebaauw, 2024-02-03:
With the DDFs I created, deCONZ shouldn't be polling the _On/Off_ cluster.  The DDF should setup attribute reporting for 0xFC03, with polling of that cluster as fallback.

Is the bulb on the latest firmware?

Last time I checked, the Hue bridge still doesn't setup nor use attribute reporting, and also polls _On/Off_, _Level Control_, and _Color Control_.  The attributes from these clusters should still reflect the light state (and even support attribute reporting).

FlatTV, 2024-02-03:
LTW001 has FW 67.88.1, that’s the last version

ebaauw, 2024-02-04:
I think not, see https://www.philips-hue.com/en-us/support/release-notes/lamps

FlatTV, 2024-02-05:
LTW001 now has 67.108.5 with same effect.

ebaauw, 2024-02-05:
Does the faulty on state disappear:
- When reading the _On/Off_ cluster attributes from the GUi?
- When reading the _Hue Effects_ cluster attributes from the GUI?

After turning the light off through the API, does the state change:
- When reading the _On/Off_ cluster attributes from the GUi?
- When reading the _Hue Effects_ cluster attributes from the GUI?

We assumed the LTW001 would behave like the LTW010 and LTW012, but it does have adifferent firmware image (0x0104) from the other two (0x010C).  Maybe the LTW001 implements the _Hue Effects_ cluster (0xFC03) differently?  Do you see any errors or warning when turning on DDF and JavaScript debugging?

ebaauw, 2024-02-05:
I'm having a hard time dissecting the log.  I take it the NWK address of the LTW001 is 0x1E51?

```
14:46:04:842 APS-DATA.indication srcAddr: 0x1E51, srcEp: 0x0B dstAddrMode: 2, profile: 0x0104, cluster: 0xFC03, lqi: 255, rssi: -67
14:46:04:842 	asdu: 1c0b1033010200004106070000fec601
```
This looks like an as yet unsupported mode, 0x0700, carrying `on`, `bri`, and `ct`.
```
               attr st dt  l mode on br   ct
1c 0b 10 18 01 0200 00 41 06 0700 00 fe c601
                                 off 254 454
```
Not sure of I missed this before, or if Signify changed this in later firmware.  Previously, I saw mode 0x000F, carrying `on`, `bri`, `ct`, anf `xy` (computed from ct).

Normally, the script would ignore this unknown mode, but since it won't set any item, the (legacy?) code might decide to fallback to polling the _On/Off_ cluster?

The good news: I'm also seeing the mode on my Hue Color Temperature and Extended Color lights (both ZLL and ZB3), after _Move to Color Temperature_.  I'll update `fc03_state.js` to handle this.  Still not sure if that actually causes your issue, though.

closed 2024-02-07 by manup

FlatTV, 2024-02-12:
This commit fix the problem [https://github.com/dresden-elektronik/deconz-rest-plugin/commit/b8cb3b3ebeb05c031efee58e2aefaa66c095877a](url)