dresden-elektronik/deconz-rest-plugin #7553: Boosch Room Thermostat II 230V (RBSH-RTH0-ZB-EU)

aSauerwein, 2024-01-28:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Boosch Room Thermostat II 230V (RBSH-RTH0-ZB-EU)

### Manufacturer

Bosch

### Model identifier

RBSH-RTH0-ZB-EU

### Device type to add

Thermostat

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat
I'd like to add a screenshot but is it somehow possible to enlarge the table in deconz-gui ? 
currently it shows only 4 attributes without scrolling
[image]

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

aSauerwein, 2024-01-28:
for testing I copied https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/bosch/room_thermostat2.json, changed the model id and imported it into deconz. 
So far it looks good, i can read and change the temperature setpoint. 

So I'd assume most of the things can be copied from https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/bosch/room_thermostat2.json. 

also according the the implemention from zigbe2mqtt it looks very similar
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/bosch.ts

Mimiix, 2024-01-28:
Can you please add all screenshots?

aSauerwein, 2024-01-29:
is there a way to make one screenshot of the whole attribute table ? currently it shows only 4 elements without scrolling

Mimiix, 2024-01-29:
> is there a way to make one screenshot of the whole attribute table ? currently it shows only 4 elements without scrolling

No, not that I know of.  Perhaps you can increase your resolution.

aSauerwein, 2024-02-02:
just updated the Issue with all screenshots. Let me know if you need more

aSauerwein, 2024-02-02:
I'm afraid to make a pull request, cause I just copy pasted things from the existing battery power room thermostat. 
but this ddf seems to work fine for me: 
https://gist.github.com/aSauerwein/b5ce7460a78820c8bb6dcdc451118a10

changes: 
* i removed the config/battery
* added state/valve with read&parse of 0x4020

open questions
* I did not understand what the "bindings" are used for
* deconz-gui still shows the battery icon. don't know what needs to be changed to get that icon removed. 
* for state/valve I just used `Item.val = Attr.val` but I'm not sure if that correctly matches the percent valve opening
* I cannot control the thermostat with phoscon web app. It just displays 40F and nothing changes, but it's working fine over homeassistant and rest api

SwoopX, 2024-02-03:
Why are you afraid of making a PR? Nobody was ever eaten alive as far as I recall 🙂

Just be brave!

closed 2024-03-04 by github-actions[bot]