dresden-elektronik/deconz-rest-plugin #7554: deConz 2.25.1 on macOS don’t start with connected ConBee II

RP-Develop, 2024-01-29:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

deConz 2.25.1 freezes when I try to start it with a ConBee II plugged in. If I start deConz without a plugged Conbee and plug it in after start, than works it.
It isn’t usable, because I cannot reboot the Mac, which works as a server.

Note: it works with a ConBee III, but I have other problems with ConBee III, like the missing of some nodes after reboot and that the batteries of any devices goes down very fast.

### Steps to reproduce the behavior

1. plug in a ConBee II
2. Start deConz
3. deConce is freezed

1. unplug a ConBee II
2. Start deConz
3. plug in the ConBee II
4. after some seconds is showing the network

### Expected behavior

deConz should start with a connected ConBee II

### Environment

- Host system: macOS 14.3 (23D56)
- Running method: macOS App runs with Rosetta
- Firmware version: 26720700, 26780700
- deCONZ 2.25.1
- Device: ConBee II 
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: USB Stick

SwoopX, 2024-02-02:
Confirmed and not only exclusive to MacOS

manup, 2024-02-04:
I fail to reproduce the problem on a Raspberry Pi 3B with deCONZ v2.25.1, here it starts just fine with the ConBee II already attached :thinking:

manup, 2024-02-05:
On macOS I could reproduce the issue and added a small fix.
Problem was that /dev/tty.xxxx paths don't load and instead only /dev/cu.xxxxx paths are used.

SwoopX, 2024-02-05:
> I fail to reproduce the problem on a Raspberry Pi 3B with deCONZ v2.25.1, here it starts just fine with the ConBee II already attached :thinking: 

@manup sorry, my fault. The comment was intended for a different issue, apologies for the inconvenience.

dkim151-seneca, 2024-04-17:
Happening with mine as well. I have to unplug conbee2 then start the app and then plug it back in. happens on my intel based Mac mini with Sonoma 14.4.1 installed