dresden-elektronik/deconz-rest-plugin #7558: ConBee III generates increased energy consumption in some devices

RP-Develop, 2024-01-29:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

ConBee III with firmware 264f0900 generates increased energy consumption in some devices. The battery was empty after approx. 24...40h. I observe it explicitly with:
- SYMFONISK Sound Controller E1744
- TRADFRI on/off switch E1743
- TRADFRI SHORTCUT Button E1812

After replacing the battery, it was not possible to reactivate it. The button events could not be seen in the INFO log. It was necessary to delete the device and teach it in again.

It looks much better with a ConBee II.

### Steps to reproduce the behavior

- wait until the battery is empty and the device don't send any events
- exchange the battery and try to send a button event

### Expected behavior

The energy consumption should be such that the battery lasts longer than just a few days.

### Environment

- Host system: macOS 14.3
- Running method: deConz App, runs with Rosetta
- Firmware version: 264f0900
- deCONZ version (not Home assistant Addon version!): 2.25.1
- Device: ConBee III 
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: USB Stick

sieren, 2024-01-31:
Thanks for filing this, was about to do the same.
Host system: Raspbian OS
Running method: deConz App in Docker
Firmware version: 264f0900
deCONZ version (not Home assistant Addon version!): 2.24.3
Device: ConBee III
Do you use an USB extension cable: yes
Is there any other USB or serial devices connected to the host system? No

UBEXHH, 2024-02-06:
Hence this reaction time was my solution: Return the Conbee 3 to Amazon for a refund

SwoopX, 2024-02-10:
@RP-Develop There was a new CB3 firmware release addressing quite a number of issues. You also might want to check out if it addresses this topic.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Firmware-Changelog

sieren, 2024-02-11:
Seems fixed for me.

closed 2024-03-12 by github-actions[bot]