dresden-elektronik/deconz-rest-plugin #7564: Hiking Single Phase 65A DIN Rail Energy Meter (modelid: TS0601, manufacturer: _TZE204_bkkmqmyo)

maltekluever, 2024-02-01:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Hiking/Tuya Zigbee Single Phase 65A DIN Rail Energy Meter

### Manufacturer

_TZE204_bkkmqmyo

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

[image]

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

[image]

[image]

[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

### Any other cluster of relevance/interest
**Groups**
[image]

[image]

[image]

**Scenes**
[image]

[image]

[image]

[image]

[image]
 
**Tuya Specific**
[image]

**OTAU**
[image]

[image]

[image]

**Time**
[image]

maltekluever, 2024-02-24:
push

maltekluever, 2024-02-27:
Thanks a lot for your comments! If I understand correctly, the following DDF should work:
`
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_lsanae15", "_TZE200_bkkmqmyo", "_TZE200_eaac7dkw", "_TZE204_bkkmqmyo"],
  "modelid": ["TS0601", "TS0601", "TS0601", "TS0601"],
  "product": "Single Phase Din Rail Smart Energy Meter",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
      {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val * 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "const buffer = Buffer.from(Attr.val, 'hex'); Item.val = (buffer[4] | buffer[8] << 8) / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "const buffer = Buffer.from(Attr.val, 'hex'); Item.val = (buffer[1] | buffer[0] << 8) / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 103, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
`

I saved this json to a file "hiking.json".
If I edit the ddf via deconz (right click on the device > Edit DDF > Load the file "hiking.json" > "Hot Reload"), deconz shows me the correct json in preview and it seems to have it updated. If I close the editing-window though and right click > Edit DDF again on the device, the old json is there. It seems that somehow the changes were not saved. Am I overseeing something?

Smanar, 2024-02-28:
> If I edit the ddf via deconz (right click on the device > Edit DDF > Load the file "hiking.json" > "Hot Reload"), deconz shows me the correct json in preview and it seems to have it updated. If I [...]

Honneslty I don't fight with editor, so much problem.
Just create a file in a "devices" folder and use text editor.
what is your OS ? (To know the "devices" folder to use)

Else the result in the API don't look so bad, ecxept current/voltage can be not working, but power need to work.

maltekluever, 2024-03-01:
I am running deconz on Raspbian, I guess the folder is then
`/home/pi/.local/share/dresden-elektronik/deCONZ/devices` ?

In there I created a file `hiking.json`, pasted the following content in it:
```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_lsanae15", "_TZE200_bkkmqmyo", "_TZE200_eaac7dkw", "_TZE204_bkkmqmyo"],
  "modelid": ["TS0601", "TS0601", "TS0601", "TS0601"],
  "product": "Single Phase Din Rail Smart Energy Meter",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
      {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val * 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "const buffer = Buffer.from(Attr.val, 'hex'); Item.val = (buffer[4] | buffer[8] << 8) / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "const buffer = Buffer.from(Attr.val, 'hex'); Item.val = (buffer[1] | buffer[0] << 8) / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 103, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```
Then I restarted deconz and voilà, by right-clicking on the device and "Edit DDF" it shows now the proper DDF. Thanks for the hint.

However, in the API it looks now like this:
[image]

Regarding the logs I restarted phoscon and immediately open the debug view and waited for some time. I got this, but nothing with TYDATA:

[image]

Smanar, 2024-03-02:
>I am running deconz on Raspbian, I guess the folder is then
>/home/pi/.local/share/dresden-elektronik/deCONZ/devices ?

Yes, for exemple, you are using the "user" folder so the file will be never removed.
You will just have some issue on `attr/swversion` because the DDF use an external file that is not present for you, but not a problem for the moment.

>However, in the API it looks now like this:

You have only this one ?
You need to have 3 devices created with this DDF.

And you have tuya log from legacy core instead of DDF core (and from another device, you can compare the MAC adress), I think something have failed, now the DDF is correctly set, perhaps with deleting and re-including the device ?
To work this device need the "tuya unlock sequence" and the DDF do it only at inclusion and every 24h.

Smanar, 2024-03-03:
Don't take care about the "line", when a device is offline, the bname is grayed.
It's a long story, but the line don't mean "connexion".

The tuya_unlock sequence is something invisible for you.

```
         {
          "name": "config/tuya_unlock",
          "public": false,
          "refresh.interval": 86400,
          "default": false
        },
```

You haven't saved ?
Because I suspect the editor corrupt DDF, if you take a look in the DDF on this topic there is just 

```
        {
          "name": "config/tuya_unlock"
        },
```

You can use it to take a look, but never save the change, always use a text editor to edit DDF.

Still nothing in logs with "info" ? for the device 70:ac:08:ff:fe:a1:65:fd ?

maltekluever, 2024-03-05:
> Don't take care about the "line", when a device is offline, the bname is grayed. It's a long story, but the line don't mean "connexion".

okay, thanks for the clarification!
 
> The tuya_unlock sequence is something invisible for you.
>
> ```
>          {
>           "name": "config/tuya_unlock",
>           "public": false,
>           "refresh.interval": 86400,
>           "default": [...]

yep, in the DDF I saved in the corresponding folder only the `{"name": "config/tuya_unlock" }`. However, if I right-click on the DDF and check the Preview, then I see:
```json
{
"name": "config/tuya_unlock",
"public": false,
"refresh.interval": 86400,
"default": false
},
```

> Still nothing in logs with "info" ? for the device 70:ac:08:ff:fe:a1:65:fd ?

No, unfortunately not. I'll keep the debug-log-mode on for some days and see if something pop ups there.

Smanar, 2024-03-06:
>However, if I right-click on the DDF and check the Preview, then I see:

Strange, but why not, the editor is something alive ^^.
But the path displayed on the editor title is the good one ?
The editor haven't changed other thing ?

I have checked again on Z2M project, and there is nothing special for this device, except the tuya unlock sequence
With the falg "Info" you need to have something like

21:44:58:632 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003
21:44:58:633 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003
21:44:58:634 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003
21:44:58:635 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003
21:44:58:637 TY_DATA_REPORT: seq 66, dpid: 0x69, type: 0x02, length: 4, val: 5003

And don't need to wait 24h to have them.

For me the problem is more on the zigbee side.
You have tried to delete and re-include it ?

maltekluever, 2024-03-18:
Sorry for the delayed answer, I needed an electrician to do some work so I can test it again. 

And indeed, simply deleting it and pairing it again did the trick. Thank you so much!!! :) 

Now it works like a charm and I can see it like this:
[image]

Just to sum up, the following DDF (put into the file "~/.local/share/dresden-elektronik/deCONZ/devices/hiking.json) worked:
```json
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZE200_lsanae15", "_TZE200_bkkmqmyo", "_TZE200_eaac7dkw", "_TZE204_bkkmqmyo"],
  "modelid": ["TS0601", "TS0601", "TS0601", "TS0601"],
  "product": "Single Phase Din Rail Smart Energy Meter",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
      {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 16, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 16, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "tuya"}
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val * 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "none"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/current",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "const buffer = Buffer.from(Attr.val, 'hex'); Item.val = (buffer[4] | buffer[8] << 8) / 1000;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/voltage",
          "parse": {"fn": "tuya", "dpid": 6, "eval": "const buffer = Buffer.from(Attr.val, 'hex'); Item.val = (buffer[1] | buffer[0] << 8) / 10;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "state/power",
          "parse": {"fn": "tuya", "dpid": 103, "eval": "Item.val = Attr.val / 10;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
  ]
}
```

Smanar, 2024-03-18:
Ok so  nothing to do ? Just adding your model to the list ?
Do you want to make the PR to submit the device officialy or want I make it ?

closed 2024-04-16 by github-actions[bot]