dresden-elektronik/deconz-rest-plugin #7566: Human Presence Detector - ZY-M100-S - not working properly

DIYtechie, 2024-02-02:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

    Product name: Human Presence Detector - ZY-M100-S
    Manufacturer: _TZE200_ztc6ggyl
    Model identifier: TS0601
    Device type : Sensor

Just installed the deCONZ add-on in Home Assistant version 6.22 (2.22.02). As I read the release notes, manually adding a DDF should no longer be necessary for this device, correct? I see no DFFs in the folder: /data/core_deconz/.local/share/dresden-elektronik/deCONZ/devices

[image]

And when I add the sensor, it now (mis)behaves as it did before I added the DFF in my old setup (separate docker container - not running as add-on) (cf. my old comment in this thread: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6517#issuecomment-1554318835).

In other words, the current behaivior is that it is spamming my zigbee network with updates every second and sends false negative shortly after detecting motion. What to do? The default DFF appears to be set to gold:

[image]

I tried deleting and re-pairing without it solving the issue.

### Steps to reproduce the behavior

Add the sensor via the deCONZ add-on in Home Assistant.

### Expected behavior

 Keep detecting presence as long as someone is in the room. Not spamming the zigbee network constantly. On the non-add-on version of deCONZ the sensor was working properly after adding the DDF.

### Environment

- Host system: (Raspberry Pi 4 (GB w 256 GB SSD)
- Running method: (Home Assistent deCONZ Add-on)
- Firmware version: (26720700)
- deCONZ version (not Home assistant Addon version!): (2.22.02)
- Device: (ConBee II)
- Do you use an USB extension cable: (yes) 
- Is there any other USB or serial devices connected to the host system? Yes, USB SSD

BabaIsYou, 2024-02-02:
It seems that you're not using the latest DDF because item "config/duration" has been replaced by "config/fadingtime" to avoid this kind of behavior (https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/ZY-M100_human_breathing_presence.json)

DIYtechie, 2024-02-02:
But do I need to manually edit the DDF, when I’m running the latest version of deconz? Isn’t it part of the package?

(Changing/adding DDF via the add-on seems complicated and risky)

BabaIsYou, 2024-02-05:
> But do I need to manually edit the DDF, when I’m running the latest version of deconz? Isn’t it part of the package?

You don't if you have the version containing the fix (2.24 and above). But you have version 2.22 installed.

DIYtechie, 2024-02-06:
> > But do I need to manually edit the DDF, when I’m running the latest version of deconz? Isn’t it part of the package?
> 
> You don't if you have the version containing the fix (2.24 and above). But you have version 2.22 installed.

Any idea when that version will be available to deCONZ add-on users?

DIYtechie, 2024-02-19:
New version 6.23.0 (add-on) solved this issue. Thanks!

closed 2024-02-19 by DIYtechie

brientim, 2024-11-30:
The WZ-M100 is an identical clone of the ZY-M100 and has the same capabilities and functions. My question is rather than raise a new devices request, it would be more appropriate to add the _TZE204_laokfqwu to the existing MmwRadar Human Presence Detector (TS0601) JSON or should I raise a new device request to WZ_M100_mmwave_radar?

Product name: Human Presence Detector - WY-M100-S
[WZ_M100_mmwave_radar.json](https://github.com/user-attachments/files/17967147/WZ_M100_mmwave_radar.json)

Manufacturer: _TZE204_laokfqwu
Model identifier: TS0601
Device type : Sensor

msebald, 2025-01-23:
@brientim does your WZ-M100 device work with this JSON file?

I also bought such a device, but it is detected as _TZE200_ya4ft0w4 and does not work with the JSON file (I edited the manufactor name and the DDF is loaded).

It is detected as light (on/off switch) and also as presence and light sensors. But presence shows unknown error in the deCONZ database.

What can I do to adjust the file for my variant of this sensor?

brientim, 2025-01-24:
Noting I am running Linux and the file system may be different. 

1.  I create the file to be use and named it specific for the device i.e. _TZE204_laokfqwu_mmwave_radar.json  
2. I manually uploaded the file to folder dresden-electronik/deCONZ/devices and note that the alternative  folder maybe /deCONZ/devices/tuya.
3. in deCONZ UI right click on the and selected edit DDF (Crtl+E) 
4. Verify the Manufacture Name, Model, Vendor and Product  are correct
5. From the File Menu > Open, select the applicable DDF json file and then hot reload. 
6. I read that hot reload may not be working so you may need to restart manually 

I had to manually reload before it starting working.

Context of my action were changes:
"manufacturername": "_TZE204_laokfqwu", to "manufacturername": "_TZE200_ya4ft0w4",
Rename to
_TZE200_ya4ft0w4_mmwave_radar.json  

[_TZE204_laokfqwu_mmwave_radar.json](https://github.com/user-attachments/files/18528688/_TZE204_laokfqwu_mmwave_radar.json)