dresden-elektronik/deconz-rest-plugin #7573: Aqara Smart Plug with deconz reports jumping energy metering

Haves1001, 2024-02-03:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I bought and added the plug (lumi.plug.maeu01) and paired it with deCONZ. After a reboot all desired attributes are shown, but unfortunately the consumption of the device is always jumping between a base value (0 in the beginning, seems to set to higher values after a reboot) and the real value.

### Steps to reproduce the behavior

Pair the Aqara Smart Plug via the deCONZ UI
Restart Home Assistant to make the consumption sensor to show up
Use the consumption sensor for the energy dashboard

### Expected behavior

The plug sensor in HomeAssistant should report values only in an increasing manner without jumping back to lower values.

### Screenshots

[image]
[image]
[image]
As a result the energy dashboard is totally off, as this sensor is seen as going up and down and therefore causing issues with accumulating the consumption values.
[image]

### Environment

- Host system: Hassio Raspberry Pi 4
- Running method: Home Assistent deCONZ Add-on
- Firmware version: 26720700
- deCONZ version (not Home assistant Addon version!): 2.22.02 / 19.9.2022
- Device: ConBee II
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system?: no

### deCONZ Logs

16:02:10:392 	03 Device temperature 32 °C
16:02:10:393 	98 power 106.127998 (106)
16:02:10:394 	95 consumption 2.465780 (2466)
16:02:10:395 	96 voltage 2280.000000 (228)
16:02:10:396 	97 current 465.473694 (465)
16:02:10:396 	05 RSSI dB (?) 3 (0x0003)
16:02:10:397 	9a unknown 0 (0x00)
16:02:10:398 	08 unknown 297 (0x0129)
16:02:10:398 	09 unknown 3840 (0x0F00)
16:02:10:399 	0b unknown 0 (0x00)
16:02:10:400 	9b Consumer connected (yes/no) 1
16:02:10:400 	0a Parent NWK 31099 (0x797B)
16:02:10:401 	0c unknown 1 (0x01)
16:02:10:402 ZCL attribute report 0x54EF4410007EDC94 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
16:02:23:664 ZCL attribute report 0x003C84FFFEDC4CAA for cluster: 0x0405, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
16:02:24:092 0x54ef44100062af7f found group 0xFFF0
16:02:44:224 0x54EF4410007EEECB extract Xiaomi special attribute 0x00F7
16:02:44:225 	64 on/off 1
16:02:44:226 	03 Device temperature 29 °C
16:02:44:228 	98 power 0.751000 (1)
16:02:44:229 	95 consumption 0.295945 (296)
16:02:44:230 	96 voltage 2290.000000 (229)
16:02:44:231 	97 current 3.279476 (3)
16:02:44:231 	05 RSSI dB (?) 7 (0x0007)
16:02:44:233 	9a unknown 0 (0x00)
16:02:44:234 	08 unknown 297 (0x0129)
16:02:44:234 	09 unknown 3840 (0x0F00)
16:02:44:235 	0b unknown 0 (0x00)
16:02:44:235 	9b Consumer connected (yes/no) 1
16:02:44:236 	0a Parent NWK 0 (0x0000)
16:02:44:236 	0c unknown 1 (0x01)
16:02:44:237 ZCL attribute report 0x54EF4410007EEECB for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
16:03:47:557 ZCL attribute report 0x00158D0009464728 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
16:03:47:559 0x00158D0009464728 extract Xiaomi special attribute 0xFF01
16:03:47:560 	01 battery 3105 (0x0C21)
16:03:47:561 	03 Device temperature 24 °C
16:03:47:562 	04 unknown 5032 (0x13A8)
16:03:47:563 	05 RSSI dB (?) 12 (0x000C)
16:03:47:564 	06 LQI (?) 4294967296 (0x0100000000)
16:03:47:566 	0a Parent NWK 18621 (0x48BD)
16:03:49:523 0x54EF4410007EEC28 extract Xiaomi special attribute 0x00F7
16:03:49:525 	64 on/off 1
16:03:49:526 	03 Device temperature 32 °C
16:03:49:527 	98 power 0.387000 (0)
16:03:49:528 	95 consumption 3.674345 (3674)
16:03:49:529 	96 voltage 2250.000000 (225)
16:03:49:530 	97 current 1.720000 (2)
16:03:49:531 	05 RSSI dB (?) 1 (0x0001)
16:03:49:531 	9a unknown 0 (0x00)
16:03:49:532 	08 unknown 297 (0x0129)
16:03:49:533 	09 unknown 3840 (0x0F00)
16:03:49:534 	0b unknown 0 (0x00)
16:03:49:534 	9b Consumer connected (yes/no) 1
16:03:49:535 	0a Parent NWK 31099 (0x797B)
16:03:49:536 	0c unknown 1 (0x01)
16:03:49:538 ZCL attribute report 0x54EF4410007EEC28 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
16:04:03:959 ZCL attribute report 0x54EF4410007EDB75 for cluster: 0xFCC0, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
DJS_GetResourceEndpoints
DJS_GetResourceEndpoints
DJS_GetResourceEndpoints
DJS_GetResourceEndpoints
16:04:21:857 Bind response success for 0x54ef44100062af7f ep: 0x01 cluster: 0x0006
16:04:21:858 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x54EF44100062AF7F (seems to be active)
16:04:55:789 Skip idle timer callback, too early: elapsed 913 msec
16:04:57:792 GW firmware version is up to date: 0x26720700
16:04:58:407 0x54EF44100062AF7F extract Xiaomi special attribute 0x00F7
16:04:58:408 	64 on/off 1
16:04:58:409 	03 Device temperature 31 °C
16:04:58:413 	98 power 1.266000 (1)
16:04:58:414 	95 consumption 0.690714 (691)
16:04:58:415 	96 voltage 2280.000000 (228)
16:04:58:415 	97 current 5.552632 (6)
16:04:58:416 	05 RSSI dB (?) 1 (0x0001)
16:04:58:417 	9a unknown 0 (0x00)
16:04:58:417 	08 unknown 299 (0x012B)
16:04:58:418 	09 unknown 3328 (0x0D00)
16:04:58:419 	0b unknown 0 (0x00)
16:04:58:419 	9b Consumer connected (yes/no) 1
16:04:58:420 	0a Parent NWK 18621 (0x48BD)
16:04:58:421 	0c unknown 1 (0x01)
16:04:58:422 ZCL attribute report 0x54EF44100062AF7F for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
16:05:23:980 ZCL attribute report 0x54EF4410007627CB for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
16:05:24:090 ZCL attribute report 0x54EF4410007627CB for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
16:05:59:788 Skip idle timer callback, too early: elapsed 867 msec
16:06:23:934 0x54EF44100070D147 extract Xiaomi special attribute 0x00F7
16:06:23:935 	64 on/off 1
16:06:23:936 	03 Device temperature 33 °C
16:06:23:937 	98 power 74.797997 (75)
16:06:23:938 	95 consumption 2.073241 (2073)
16:06:23:939 	96 voltage 2180.000000 (218)
16:06:23:940 	97 current 343.110077 (343)
16:06:23:941 	05 RSSI dB (?) 7 (0x0007)
16:06:23:942 	9a unknown 0 (0x00)
16:06:23:943 	08 unknown 297 (0x0129)
16:06:23:943 	09 unknown 3072 (0x0C00)
16:06:23:944 	0b unknown 0 (0x00)
16:06:23:945 	9b Consumer connected (yes/no) 1
16:06:23:946 	0a Parent NWK 26533 (0x67A5)
16:06:23:947 	0c unknown 1 (0x01)
16:06:23:949 ZCL attribute report 0x54EF44100070D147 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F

Haves1001, 2024-02-25:
Hi @SwoopX I saw you working on a similar issue, would you be able to take a look at this one?

Haves1001, 2024-03-20:
Any chance that anyone could take a look at this bug?

Mimiix, 2024-04-11:
@SwoopX you know these, perhaps you can see some light?

SwoopX, 2024-04-12:
Maybe this? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-smart-plugs-not-working-correctly

And the issue is a duplicate

closed 2024-05-16 by github-actions[bot]