dresden-elektronik/deconz-rest-plugin #7587: Tuya Smoke Detector YG400A - DDF needed

Siggi0904, 2024-02-10:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya Smoke Detector YG400A

### Manufacturer

Tuya

### Model identifier

YG400A

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

[image]

### Basic

[image]

### Further relevant clusters

Please cross check to TS0205 in addition a sirene.
See too: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3557 at version 2.6.0 release.

Forum: https://forum.phoscon.de/t/tuya-smoke-detector-yg400a-ddf-needed/4534/16

Smanar, 2024-02-10:
Hello, can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_up3pngle",
  "modelid": "TS0205",
  "vendor": "Tuya",
  "product": "Smoke Sensor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/test"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 600,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

Siggi0904, 2024-02-10:
Hi @Smanar, is the modelname correct? 

Can I test activate the sirene, like in iobroker?

Great thanks.

Smanar, 2024-02-10:
```
  "manufacturername": "_TZ3210_up3pngle",
  "modelid": "TS0205",
```
Yep this is from the capture, so from the device itself, it's tuya, they can write anything on the box as model, even on different hardware.

Siggi0904, 2024-02-11:
@Smanar looks good, could be added for next release.
I called the json file: _TZ3210_TS0205_YG400A_smoke_sensor.json

In Phoscon App:

[image]

In IOBroker:

[image]

Smanar, 2024-02-11:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7593

Siggi0904, 2024-02-22:
@Smanar can you check the ddf file for updating device connection?
In phoscon the device is shown with long time away for last updated.

With this value, phoscon shows the device as not reachable after around 1 week.

Smanar, 2024-02-22:
It's not something normal ?
Can you show the device json from for exemple phoscon/help/API Information/sensor ?
The battery report need to be enought I think.

Siggi0904, 2024-02-22:
`{
    "config": {
        "battery": 100,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 1,
    "etag": "2f8975ceabfee16ba425f0623f6cf864",
    "lastannounced": "2024-02-11T11:41:59Z",
    "lastseen": "2024-02-22T17:15Z",
    "manufacturername": "_TZ3210_up3pngle",
    "modelid": "TS0205",
    "name": "Rauchmelder Wohnzimmer",
    "state": {
        "fire": false,
        "lastupdated": "2024-02-20T17:10:30.924",
        "lowbattery": false,
        "test": false
    },
    "swversion": "1.0.1",
    "type": "ZHAFire",
    "uniqueid": "a4:c1:38:a3:67:11:fc:c2-01-0500"
}`

but phoscon says last seen 2 days ago:
[image]

Smanar, 2024-02-22:
So for me all is fine.
`"lastseen": "2024-02-22T17:15Z"`
Not the device fault if you haven't fire in your home and the value have not moved ^^

Siggi0904, 2024-02-24:
okay, will have a look at this device, if it's not reachable.

closed 2024-03-24 by github-actions[bot]

BooWseR, 2024-04-26:
Hi there,
my smoke alarms arrived today. I tried it with this DDF, but unfortunately some values are missing:
[image]
`{
    "config": {
        "battery": 0,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 1,
    "etag": "9ca8e8bd66b25e45f614d88eb668322c",
    "lastannounced": "2024-04-26T19:25:17Z",
    "lastseen": "2024-04-26T19:24Z",
    "manufacturername": "_TZ3210_up3pngle",
    "modelid": "TS0205",
    "name": "TS0205",
    "state": {
        "fire": null,
        "lastupdated": "none",
        "lowbattery": null,
        "test": false
    },
    "type": "ZHAFire",
    "uniqueid": "a4:c1:38:ca:d8:b9:00:ad-01-0500"
}`

Can anyone help? Thanks in advance.

Smanar, 2024-04-27:
I don't see something special for this device, the enrollement have probably failed.
Can work after a time, or try to  re-include the device.

If you have the GUI, IAS debug can help.