dresden-elektronik/deconz-rest-plugin #7619: Tuya Scene Switch TS0044 TZ3000_a4xycprs

wunni1, 2024-02-17:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

MOES ZigBee Scene Switch

### Manufacturer

TZ3000_a4xycprs

### Model identifier

TS0044

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off
[image]
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2024-02-18:
Hello can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "TZ3000_a4xycprs",
  "modelid": "TS0044",
  "product": "Tuya remote 4 gangs locked version",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0001",
            "at": "0x0021"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tuya_unlock"
        }
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 4,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
```

IDK how will react battery return ...

Remember you need too add the Manufacture Name in the button_maps.json file

```
        "Tuya3gangMap": {
            "vendor": "Tuya",
            "doc": "3-gang remote",
            "modelids": ["_TZ3000_yw5tvzsk", "_TZ3000_czuyt8lz","_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_4upl1fcj", "_TZ3000_i3rjdrwu", "_TZ3000_w3c7ouru", "_TZ3000_abci1hiu","_TZ3000_wqcbzbae", "_TZ3000_pkeqinnt", "_TZ3000_uaa99arv", "_TZ3000_wbfgbpxq", "_TZ3000_fa9mlvja", "_TZ3000_tk3s5tyg", "_TZ3000_mrpevh8p", "_TZ3000_qgwcxxws"],
            "map": [
```

wunni1, 2024-02-19:
Thank you! But unfortunately i can’t find the file/folder according to the manuals I’ve found. Running DeConz as HASS Addon with an RPI4.

Smanar, 2024-02-19:
Ha bad luck, it's possible to add new DDF > https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839
But not possible to edit the button_maps.json file.

wunni1, 2024-02-19:
This is exactly what I tried... :-(

Smanar, 2024-02-19:
It's an hidden folder
`/data/.local/share/dresden-elektronik/deCONZ/devices/`

But to check if it's working you will need to use log as you can't edit the buttonmap file. Do you have access to deconz log on application ?

wunni1, 2024-02-20:
Can’t find that folder. Access to the log is possible. Is there another way with the DDF?

Smanar, 2024-02-20:
Not on the HA docker :(
But this folder is invisible, just try to access it.

wunni1, 2024-03-10:
@Smanar : Sorry I’ve missed your answer…
Tried again, but can’t access the folder. Could you make the PR “in blind mode”. Then I will try it out. Thank you!

wunni1, 2024-05-16:
I'd like to open this issue again. Tried to upload the DDF mentioned from Smanar 18.02.24 with the new DDF upload feature. The switch is recognized as switch with 4 different buttons, but when I press the buttons nothing is happen. 

I copied this text:

``
{
  "schema": "devcap1.schema.json",
  "manufacturername": "TZ3000_a4xycprs",
  "modelid": "TS0044",
  "product": "Tuya remote 4 gangs locked version",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0006"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001",
            "script": "tuya_swversion.js"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0000",
            "at": "0x0001"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 86400,
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0001",
            "at": "0x0021"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tuya_unlock"
        }
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 2,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 3,
      "dst.ep": 1,
      "cl": "0x0006"
    },
    {
      "bind": "unicast",
      "src.ep": 4,
      "dst.ep": 1,
      "cl": "0x0006"
    }
  ]
}
 
``
in a blank file and renamed it tuya_scene_switch.ddf

Did I something wrong?
Thanks for help
Thomas

Mimiix, 2024-05-16:
This might have to do with the Button maps not updated. @Smanar can you check?

Zehir, 2024-05-16:
@wunni1 the upload page in phoscon it for DDF Bundles, you can't use it to upload the JSON file Smanar gave you.
But the current bundle format don't support button_maps file so for now you will still have to drop the json file in the devices directory and edit the button_maps.json file

wunni1, 2024-05-17:
@Smanar thank you again. I have no opportunity to edit this file with my OS. So can you try a blind PR. And I will test it?

Smanar, 2024-05-18:
PR done here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7781
But you need the next version to have the correction on the button_maps.json file.

wunni1, 2024-05-18:
Thank you very much!

closed 2024-05-26 by manup

thetoaster, 2024-10-19:
how to integrate the device, now that it is officially also supported? It shows up in vnc mode to deconz but not in phoscon web ui when searching for a switch

Smanar, 2024-10-19:
The device is included in the API ? Visible in phoscon/help/API Information/sensors ?
Else can you share how it look in deconz ? Can check too the Manufacture Name/Model id, can be a new clone.

thetoaster, 2024-10-30:
yes, device can be seen in deconz:
[image]
couldn't find it API information. can I provide anything else?

Smanar, 2024-10-30:
Can you get the manufacture name / model id ? (they are unknow on your capture)
You have them is the basic cluster (0x0000)

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

thetoaster, 2024-10-31:
got it: 
[image]

Smanar, 2024-10-31:
`_TZ3000_5tqxpine`

You just a clone.
Need to add it, what is your OS ?

thetoaster, 2024-10-31:
I am on HAOS, currently those versions
[image]

Smanar, 2024-11-01:
Ok so not possible.
On HA it's possible to add a new DDF, but you need to edit the button_maps.json file, and that is not possible.

thetoaster, 2024-11-01:
oh sad, but the DDF would at least help to receive the signals into HA, right? 
or it means a fix is possible and I have to wait for a new release?

Smanar, 2024-11-02:
Nope, you need too edit the button_maps.json file.
And as I don't find this device on google, IDK which one clone it is, so without tests, I can do nothing.

thetoaster, 2024-11-07:
can I test with data from this device (that thread is about) or support in another way? I assume it could be one of the many "clones"

Smanar, 2024-11-07:
You can test it with installing deconz with a real OS and make a test during 5mn, for information the network setting is stored on the gateway, the devices setup on the host.

And still not possible to see wich one clone is it, there is issue on it on Z2M ATM.

To make a try need to edit for exemple devices\tuya\_TZ3000_TS0044_4gang_remote_locked.json to have
```
  "manufacturername": ["_TZ3000_a4xycprs", "_TZ3000_u3nv1jwk", "TZ3000_a4xycprs"],
  "modelid": ["TS0044", "TS0044", "TS0044"],
```

and the file button_maps.json

`            "modelids": ["_TZ3000_yw5tvzsk", "_TZ3000_czuyt8lz","_TZ3000_ee8nrt2l", "_TZ3000_ygvf9xzp", "_TZ3000_t8hzpgnd", "_TZ3000_wkai4ga5", "_TZ3000_bi6lpsew", "_TZ3400_keyjhapk", "_TYZB02_key8kk7r", "_TZ3400_keyjqthh", "_TZ3400_key8kk7r", "_TZ3000_vp6clf9d", "_TYZB02_keyjqthh", "_TZ3000_peszejy7", "_TZ3000_qzjcsmar", "_TZ3000_owgcnkrh", "_TZ3000_adkvzooy", "_TZ3000_arfwfgoa", "_TZ3000_a7ouggvs", "_TZ3000_rrjr1q0u", "_TZ3000_abci1hiu", "_TZ3000_dfgbtub0", "_TZ3000_rco1yzb1", "_TZ3000_gbm10jnj", "_TZ3000_5e235jpa", "_TZ3000_sj7jbgks", "_TZ3000_w8jwkczz", "_TZ3000_dziaict4", "_TZ3000_famkxci2", "_TZ3000_itb0omhv", "_TZ3000_tzvbimpq", "_TZ3000_mh9px7cq", "_TZ3000_b7bxojrg", "_TZ3000_ufhtxr59", "_TZ3000_4upl1fcj", "_TZ3000_i3rjdrwu", "_TZ3000_w3c7ouru", "_TZ3000_abci1hiu","_TZ3000_wqcbzbae", "_TZ3000_pkeqinnt", "_TZ3000_uaa99arv", "_TZ3000_wbfgbpxq", "_TZ3000_fa9mlvja", "_TZ3000_tk3s5tyg", "_TZ3000_mrpevh8p", "_TZ3000_qgwcxxws", "_TZ3000_yj6k7vfo", "_TZ3000_axpdxqgu", "_TZ3000_a4xycprs", "_TZ3000_u3nv1jwk", "TZ3000_a4xycprs"],`