dresden-elektronik/deconz-rest-plugin #7623: New attached TRADFRI on/off button switch 1002 switches all lights at once

binford6000, 2024-02-20:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

After loosing connection due to low battery i had to reconnect the switch again. Everything works as intended but
when pressing the ON button (event 1002) all lights are switched on.

### Steps to reproduce the behavior

1. Delete switch 
2. Reconnect switch
3. Press ON-Button to create 1002 event
4. All lights (group 0x0000) will go on

### Expected behavior

After reconnecting the switch only connected lights/groups will switch on like before.

### Environment

- Host system: Ubuntu 22 VM on Proxmox Hypervisor
- Running method: deconz Docker
- Firmware version: 26780700
- deCONZ version: (2.24.3)
- Device: ConBee II
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system: no

### deCONZ Logs

13:49:23:658 [INFO] - Button 1002 - TRADFRI on/off switch, broadcast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 18
13:49:23:668 rule event /sensors/208/state/lastupdated: 0 -> 0
13:49:23:670 trigger rule 24 - Rule PUSHDIM_ON
13:49:23:774 ZCL attribute report 0x14B457FFFE2BEB2E for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:49:23:850 ZCL attribute report 0xCCCCCCFFFE8E5827 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:49:25:090 ZCL attribute report 0x00158D00032AE944 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
13:49:25:146 delay sending request 72 dt 0 ms to 0xCCCCCCFFFE8E5827, ep: 0x01 cluster: 0x0008 onAir: 1
13:49:25:147 delay sending request 73 dt 0 ms to 0xCCCCCCFFFE8E5827, ep: 0x01 cluster: 0x0300 onAir: 1
13:49:25:170 	0xCCCCCCFFFE8E5827 force poll (2)
13:49:25:187 delay sending request 73 dt 0 ms to 0xCCCCCCFFFE8E5827, ep: 0x01 cluster: 0x0300 onAir: 1
13:49:25:231 	0xCCCCCCFFFE8E5827 force poll (2)
13:49:25:280 	0xCCCCCCFFFE8E5827 force poll (2)
13:49:25:699 ZCL attribute report 0xCCCCCCFFFE8E5827 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:49:25:777 send move to color temperature 167 to 0x00158D0005826D92
13:49:25:837 	0x00158D0005826D92 force poll (2)
13:49:25:900 	0x00158D0005826D92 force poll (2)
13:49:25:983 send move to color temperature 167 to 0x60A423FFFEC8B1C4
13:49:26:043 	0x60A423FFFEC8B1C4 force poll (2)
13:49:26:102 send move to color temperature 167 to 0x00158D000582709D
13:49:26:108 	0x60A423FFFEC8B1C4 force poll (2)
13:49:26:109 delay sending request 91 dt 0 ms to 0x00158D000582709D, ep: 0x01 cluster: 0x0300 onAir: 1
13:49:26:129 	0x00158D000582709D force poll (2)
13:49:26:189 	0x00158D000582709D force poll (2)
13:49:26:245 delay sending request 94 dt 0 ms to 0xCCCCCCFFFE8E5827, ep: 0x01 cluster: 0x0300 onAir: 1
13:49:26:268 	0xCCCCCCFFFE8E5827 force poll (2)
13:49:26:343 	0xCCCCCCFFFE8E5827 force poll (2)
13:49:29:402 ZCL attribute report 0x00158D000361DB34 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
13:49:30:121 ZCL attribute report 0x00158D000361DB34 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
13:49:30:746 ZCL attribute report 0xCCCCCCFFFE8E5827 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:49:31:112 ZCL attribute report 0x00158D000361DB34 for cluster: 0x0702, ep: 0x01, frame control: 0x18, mfcode: 0x0000

ChrisHae, 2024-02-22:
I can confirm that behaviour. The lights only change their status via Rest API not the physical lights itself.
on button switches all lights on and off button switches all lights off.
Also on deCONZ v. 2.25.1

binford6000, 2024-02-23:
Thx @Mimiix 
Solved my issue with that 👍🏻

closed 2024-02-23 by binford6000

ChrisHae, 2024-02-27:
I see this is another issue. And FYI: it is fixed in new deCONZ beta 2.26.1