dresden-elektronik/deconz-rest-plugin #7636: Sonoff ZBDongle-E (Router Firmware)

LeoeLeoeL, 2024-02-25:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Sonoff ZBDongle-E

### Manufacturer

ITEAD

### Model identifier

DONGLE-E_R

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]
Deconz believes it's a lamp.

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]
If I click on EXEC buttons I receive SUCCESS
### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

BoBiene, 2024-03-09:
I have the same. It shows up in the phoscon app as two devices:
[image]

For those hwo like to use the Sonoff ZBDongle-E as range extender in phoscon / deconz:

1. Download the router firmware from https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/tree/master/Dongle-E/Router
2. Use Chrome / Edge and flash the firmware (on windows) with https://darkxst.github.io/silabs-firmware-builder/
3. Disassemble the stick and power it on
4. start pairing in phoscon
5. press and hold the upper button on the stick (not the reset button) for 5-10 seconds and release the button
6. flasing of the led should indicated the pairing.

LeoeLeoeL, 2024-04-01:
I can see a lack of support on github in the last months.

LeoeLeoeL, 2024-04-24:
Keeping alive.

LeoeLeoeL, 2024-05-16:
Hoping for a DDF...

LeoeLeoeL, 2024-06-07:
Hope is the last to die

Mimiix, 2024-06-07:
@Smanar can you help out :)?

Smanar, 2024-06-07:
Hello, and you just want to see it as router ?
Can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "DONGLE-E_R",
  "product": "ZBDongle-E",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_RANGE_EXTENDER",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

LeoeLeoeL, 2024-06-08:
Unable to load.

[image]

[image]

Mimiix, 2024-06-09:
You can't load a json, needs to be a bundle..

@Smanar can you do a PR so we get a bundle?

Smanar, 2024-06-09:
>can you do a PR so we get a bundle?

Need to test it first, no ?

BTW have checked the file again, there is no typo inside.
You have a real OS ? Can try to put the DDF in a "device" folder.

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired

Then just restart deconz.

Zehir, 2024-06-09:
@Smanar you can open a draft PR so you get a bundle and it's not get merged.

You can also use the cli tool to generate the bundle

Mimiix, 2024-06-09:
This user can't or won't test the json, he can only use the phosocn upload.

Phoscon upload only takes bundles, no jsons. Serious flaw in my opinion,but owell.

LeoeLeoeL, 2024-06-09:
Last time to check the "Avatto-ZWSM16-2" DDF I loaded the json from DDF Editor.Now I performed the same actions. Why doesn't work this time?

Smanar, 2024-06-09:
With the DDF on the folder and after a deconz restart, it need to pick itself the DDF.
To choose them it use only
```
  "manufacturername": "SONOFF",
  "modelid": "DONGLE-E_R",
```

Perhaps you still have old entry created, just remove the device first, but I don't see something bad that can prevent the DDF be used.

LeoeLeoeL, 2024-06-09:
Are we sure everything is ok with DDFs?
I say that because I just realized I lost DDF in my OSRAM plugs. I had in a previous release.

Anyway, I can't load DDF for this range extender. I tried everything.
There is a "default" DDF  (not created by me that doesn't want to be deleted.
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "DONGLE-E_R",
  "product": "DONGLE-E_R",
  "sleeper": false,
  "status": "Draft",
  "subdevices": [
    {
      "type": "$TYPE_EXTENDED_COLOR_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "cap/color/capabilities",
          "description": "The supported color modes as bitmap.",
          "refresh.interval": 86400,
          "read": {
            "at": "0x400a",
            "cl": "0x0300",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x400a",
            "cl": "0x0300",
            "ep": 0,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "cap/color/ct/max",
          "description": "Maximum supported color temperature.",
          "refresh.interval": 86400,
[... 274 lines cut ...]
            "ep": 0,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

If I remove everything:
[image]

add the range extender:
[image]

save it:
[image]

If I make a HOT RELOD or Reboot Deconz....:
I have this again!!!!
[image]

Smanar, 2024-06-10:
What is your OS ?

>There is a "default" DDF (not created by me that doesn't want to be deleted.

Are you seeing the file ? In wich one folder ?

`sudo chmod -R 777 folder_name`

To get Full right for a file.
Can keep it, but replace the contain.

LeoeLeoeL, 2024-06-10:
Deconz runs in a  Raspberry pi.
I don't see the file. That DDF exists but I don't know why.
The difference between the DDF you created and what DDF editor makes:

"schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "DONGLE-E_R",
  "product": "ZBDongle-E",
  "sleeper": false,
  "status": "Gold",

"schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "DONGLE-E_R",
  "product": "DONGLE-E_R",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/dongle-e_r.json",

LeoeLeoeL, 2024-06-10:
Is there anything different about DDFs between 2.26.3 and 2.27.2?

2.27.2
[image]

2.26.3
[image]

As I wrote above. In the latest beta I lost some DDFs.

Ok, there is something "strange" in the latest beta.
With the last stable i loaded the DDF immediately.
[image]
Anyway, the anomalyis still present (it's seen as 2 light devices)

Mimiix, 2024-06-10:
> Is there anything different about DDFs between 2.26.3 and 2.27.2?
>
> 2.27.2 [...]

The last beta is confirmed to have issues with ddfs in general. I recommend staying on 2.26.3 for now

Smanar, 2024-06-11:
>Anyway, the anomalyis still present (it's seen as 2 light devices)

With the good DDF ? Deconz have picked the good one ? you will see the same thing in deconz, the difference will be only in the API.

>The difference between the DDF you created and what DDF editor makes:

Thoses difference are normal, but if you have only `"type": "$TYPE_RANGE_EXTENDER",` on the DDF I on't see why deconz still creating the 2 others one ...

>I don't see the file

No typo in file path ? File is not hidden.

LeoeLeoeL, 2024-06-11:
I tried both.
Anyway, the DDF wasn't OK because, after minutes, the device lost all links with the others.

Smanar, 2024-06-12:
>>the DDF wasn't OK because, after minutes, the device lost all links with the others.

The device is passive, the DDF too, it don't have impact on the device, It just have a cosmetic result (at least it need to have if working)

closed 2024-07-11 by github-actions[bot]