dresden-elektronik/deconz-rest-plugin #7639: Müller-Licht bulbs create a group 240

mattreim, 2024-02-25:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

My Müller-Licht bulbs always create a group 240(0x00F0). After deletion, the group is always created again. Also unbind in the bind dropbox and remove group in the group cluster(0004) didn't work.

My bulbs: 
2x Müller-Licht - tint LED-Reflektor GU10 white - 404006
2x Müller-Licht - tint LED-Kerzenform E14 white - 404008

### Steps to reproduce the behavior

1. Delete the group from the DB.
2. Switch on the bulbs.
3. Group is created (see screenshot)

### Expected behavior

No further group will be created.

### Screenshots

Phoscon:
[image]

DB-SQLite:
[image]

### Environment

- Host system: Rock Pi 4b
- Running method: deconz-docker
- Firmware version: 26780700
- deCONZ version v2.26.0-beta
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? no

mattreim, 2024-03-18:
Unstale...

SwoopX, 2024-03-20:
Honestly, I don't think that is a bug (of deconz). My bulbs interestingly have the same group ID available. However, I can delete it permanently via API, which is the recommended way I believe.

mattreim, 2024-03-21:
Sorry, but it's more of a botch than a fix.

mattreim, 2024-04-13:
Unstale...

closed 2024-05-13 by github-actions[bot]