dresden-elektronik/deconz-rest-plugin #7641: _TZE204_sxm7l9xa doesn't show targetdistance

patrickschaper, 2024-02-26:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

The state of `_TZE204_sxm7l9xa` doesn't show the `targetdistance` property as stated [here](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/3a8e935df3c2701bcf4e2adb52c0b5985e220e9a/devices/tuya/_TZE204_sxm7l9xa_mmwave_radar.json#L277). 

Also when looking into the app the state is missing.
[image]

I'm running **deCONZ 2.25.3** on Home Assistant with a Raspbee. 

The deconz-input node in NodeRED is also missing the `targetdistance` property: 
```JSON
{
   "payload":{
      "lastupdated":"2024-02-26T16:50:16.512",
      "presence":true
   },
   "topic":"",
   "payload_format":"__complete__",
   "payload_raw":{
      "config":{
         "delay":50,
         "on":true,
         "reachable":true,
         "sensitivity":7,
         "triggerdistance":"6"
      },
      "etag":"804bdca0e7d764de3763745388c2ac72",
      "lastannounced":null,
      "lastseen":"2024-02-26T16:50Z",
      "manufacturername":"_TZE204_sxm7l9xa",
      "modelid":"TS0601",
      "name":"Living Room Motion Sensor",
      "state":{
         "lastupdated":"2024-02-26T16:50:16.512",
         "presence":true
      },
      "swversion":"1.0.10",
      "type":"ZHAPresence",
      "uniqueid":"a4:c1:38:88:7b:0b:c6:ea-01-ef00",
      "device_type":"sensors",
      "device_id":121,
      "device_path":"sensors/uniqueid/a4:c1:38:88:7b:0b:c6:ea-01-ef00"
   },
   "payload_type":"state",
   "meta":{
      "config":{
         "delay":50,
         "on":true,
         "reachable":true,
         "sensitivity":7,
         "triggerdistance":"6"
      },
      "etag":"804bdca0e7d764de3763745388c2ac72",
      "lastannounced":null,
      "lastseen":"2024-02-26T16:50Z",
      "manufacturername":"_TZE204_sxm7l9xa",
      "modelid":"TS0601",
      "name":"Living Room Motion Sensor",
      "state":{
         "lastupdated":"2024-02-26T16:50:16.512",
         "presence":true
      },
      "swversion":"1.0.10",
      "type":"ZHAPresence",
      "uniqueid":"a4:c1:38:88:7b:0b:c6:ea-01-ef00",
      "device_type":"sensors",
      "device_id":121,
      "device_path":"sensors/uniqueid/a4:c1:38:88:7b:0b:c6:ea-01-ef00"
   },
   "meta_changed":[
      
   ],
   "_msgid":"b218fedea27be1ea"
}
```

In case it helps, here my node information
[image]

### Steps to reproduce the behavior

1. In NodeRED create a deconz-input node with the motions sensor selected
2. read the complete msg object in a debug node
3. Trigger a ZHAPresence event

### Expected behavior

The message payload should contain the targetdistance.

### Environment

- Host system: Raspberry Pi
- Running method: Home Assistent deCONZ Add-on
- Firmware version: 26400500
- deCONZ version (not Home assistant Addon version!): 2.25.3
- Device: RaspBee I
- Are there any other USB or serial devices connected to the host system?: No

### deCONZ Logs

```
REPORT: seq 89, dpid: 0x6D, type: 0x02, length: 4, val: 346
18:04:18:848 TY_DATA_REPORT: seq 89, dpid: 0x6D, type: 0x02, length: 4, val: 346
18:04:18:848 TY_DATA_REPORT: seq 89, dpid: 0x6D, type: 0x02, length: 4, val: 346
18:04:20:449 TY_DATA_REPORT: seq 90, dpid: 0x6D, type: 0x02, length: 4, val: 133
18:04:20:449 TY_DATA_REPORT: seq 90, dpid: 0x6D, type: 0x02, length: 4, val: 133
18:04:20:449 TY_DATA_REPORT: seq 90, dpid: 0x6D, type: 0x02, length: 4, val: 133
18:04:20:449 TY_DATA_REPORT: seq 90, dpid: 0x6D, type: 0x02, length: 4, val: 133
18:04:20:449 TY_DATA_REPORT: seq 90, dpid: 0x6D, type: 0x02, length: 4, val: 133
18:04:22:047 TY_DATA_REPORT: seq 91, dpid: 0x6D, type: 0x02, length: 4, val: 137
18:04:22:047 TY_DATA_REPORT: seq 91, dpid: 0x6D, type: 0x02, length: 4, val: 137
18:04:22:047 TY_DATA_REPORT: seq 91, dpid: 0x6D, type: 0x02, length: 4, val: 137
18:04:22:047 TY_DATA_REPORT: seq 91, dpid: 0x6D, type: 0x02, length: 4, val: 137
18:04:22:047 TY_DATA_REPORT: seq 91, dpid: 0x6D, type: 0x02, length: 4, val: 137
18:04:23:648 TY_DATA_REPORT: seq 92, dpid: 0x6D, type: 0x02, length: 4, val: 130
18:04:23:648 TY_DATA_REPORT: seq 92, dpid: 0x6D, type: 0x02, length: 4, val: 130
18:04:23:648 TY_DATA_REPORT: seq 92, dpid: 0x6D, type: 0x02, length: 4, val: 130
18:04:23:648 TY_DATA_REPORT: seq 92, dpid: 0x6D, type: 0x02, length: 4, val: 130
18:04:23:649 TY_DATA_REPORT: seq 92, dpid: 0x6D, type: 0x02, length: 4, val: 130
18:04:25:249 TY_DATA_REPORT: seq 93, dpid: 0x6D, type: 0x02, length: 4, val: 167
18:04:25:249 TY_DATA_REPORT: seq 93, dpid: 0x6D, type: 0x02, length: 4, val: 167
18:04:25:249 TY_DATA_REPORT: seq 93, dpid: 0x6D, type: 0x02, length: 4, val: 167
18:04:25:249 TY_DATA_REPORT: seq 93, dpid: 0x6D, type: 0x02, length: 4, val: 167
18:04:25:249 TY_DATA_REPORT: seq 93, dpid: 0x6D, type: 0x02, length: 4, val: 167
18:04:26:846 TY_DATA_REPORT: seq 94, dpid: 0x6D, type: 0x02, length: 4, val: 60
18:04:26:847 TY_DATA_REPORT: seq 94, dpid: 0x6D, type: 0x02, length: 4, val: 60
18:04:26:847 TY_DATA_REPORT: seq 94, dpid: 0x6D, type: 0x02, length: 4, val: 60
18:04:26:847 TY_DATA_REPORT: seq 94, dpid: 0x6D, type: 0x02, length: 4, val: 60
18:04:26:847 TY_DATA_REPORT: seq 94, dpid: 0x6D, type: 0x02, length: 4, val: 60
18:04:28:197 0x04CF8CDF3C8A3A17 found group 0xFFF0
18:04:28:197 0x04CF8CDF3C8A3A17 found group 0x0005
18:04:28:446 TY_DATA_REPORT: seq 95, dpid: 0x6D, type: 0x02, length: 4, val: 120
18:04:28:446 TY_DATA_REPORT: seq 95, dpid: 0x6D, type: 0x02, length: 4, val: 120
18:04:28:446 TY_DATA_REPORT: seq 95, dpid: 0x6D, type: 0x02, length: 4, val: 120
18:04:28:446 TY_DATA_REPORT: seq 95, dpid: 0x6D, type: 0x02, length: 4, val: 120
18:04:28:446 TY_DATA_REPORT: seq 95, dpid: 0x6D, type: 0x02, length: 4, val: 120
18:04:29:263 ZCL attribute report 0x00158D0001DFC078 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
18:04:36:244 TY_DATA_REPORT: seq 96, dpid: 0x69, type: 0x04, length: 1, val: 0
18:04:36:244 TY_DATA_REPORT: seq 96, dpid: 0x69, type: 0x04, length: 1, val: 0
18:04:36:244 TY_DATA_REPORT: seq 96, dpid: 0x69, type: 0x04, length: 1, val: 0
18:04:36:245 TY_DATA_REPORT: seq 96, dpid: 0x69, type: 0x04, length: 1, val: 0
18:04:36:245 TY_DATA_REPORT: seq 96, dpid: 0x69, type: 0x04, length: 1, val: 0
18:04:37:645 TY_DATA_REPORT: seq 97, dpid: 0x6D, type: 0x02, length: 4, val: 0
18:04:37:645 TY_DATA_REPORT: seq 97, dpid: 0x6D, type: 0x02, length: 4, val: 0
18:04:37:645 TY_DATA_REPORT: seq 97, dpid: 0x6D, type: 0x02, length: 4, val: 0
18:04:37:645 TY_DATA_REPORT: seq 97, dpid: 0x6D, type: 0x02, length: 4, val: 0
18:04:37:646 TY_DATA_REPORT: seq 97, dpid: 0x6D, type: 0x02, length: 4, val: 0
18:04:39:243 TY_DATA_REPORT: seq 98, dpid: 0x6D, type: 0x02, length: 4, val: 0
18:04:39:243 TY_DATA_REPORT: seq 98, dpid: 0x6D, type: 0x02, length: 4, val: 0
18:04:39:243 TY_DATA_REPORT: seq 98, dpid: 0x6D, type: 0x02, length: 4, val: 0
18:04:39:243 TY_DATA_REPORT: seq 98, dpid: 0x6D, type: 0x02, length: 4, val: 0
18:04:39:243 TY_DATA_REPORT: seq 98, dpid: 0x6D, type: 0x02, length: 4, val: 0
```

mattreim, 2024-03-19:
Try version v2.26.1, some attributes are available again.

patrickschaper, 2024-03-19:
Alright, I guess I’ll wait for home-assistant release then. Thank you.

closed 2024-03-19 by patrickschaper