dresden-elektronik/deconz-rest-plugin #7644: Lidl bulb TS0505A has lost colours

LeoeLeoeL, 2024-02-26:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

It's impossible to set an RGB colour. Only "whites" are available.

### Steps to reproduce the behavior

Move the sliders in Phoscon never show a colour.
Same behaviour in Home Assistant.

### Expected behavior

Set a colour

### Environment

- Host system: (Raspberry Pi)
- Running method: (Raspbian)
- Firmware version: (26780700)
- deCONZ version (not Home assistant Addon version!): (2.25.3)
- Device: (ConBee II)
- Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

### Additional context

For info, the lamp has the same issue in Zigbee2MQTT (using a Sonoff ZB Dongle-P).
The lamp works nomally if paired to a Hue bridge or Lidl bridge

LeoeLeoeL, 2024-03-19:
Is there anybody out there?

LeoeLeoeL, 2024-04-10:
I really would like to know if my mesh is the only one in the world that suffers this issue.

ebaauw, 2024-04-10:
Is the light exposed by DDF or by legacy code?  Please attach deCONZ GUI screenshots so the node, including endpoints and clusters, and of the _Basic _ and _Color Control_ cluster attributes, making sure to read these first.

LeoeLeoeL, 2024-04-10:
Hi, nice to see you.
Legacy code

[image]
Basic
[image]
On/Off
[image]
Level Control
[image]
Color Control
[image]
These Livarno lights worked OK until 6 months ago (or something more)

ebaauw, 2024-04-10:
What about the other attributes in _Color Control_?  Notably _Color Capabilities_.

LeoeLeoeL, 2024-04-10:
Everything else I didn't show is greyed.
{
    "capabilities": {
        "alerts": [
            "none",
            "select",
            "lselect"
        ],
        "color": {
            "ct": {
                "max": 65279,
                "min": 0
            },
            "modes": []
        }
    },
    "colorcapabilities": 0,
    "config": {
        "groups": [
            "0",
            "3"
        ]
    },
    "ctmax": 65279,
    "ctmin": 0,
    "etag": "610fa80cab1ba813d2fb2abc71a3a9a4",
    "hascolor": true,
    "lastannounced": "2024-03-21T01:11:38Z",
    "lastseen": "2024-04-10T15:26Z",
    "manufacturername": "_TZ3000_dbou1ap4",
    "modelid": "TS0505A",
    "name": "Extended color light 7",
    "state": {
        "alert": "none",
        "bri": 255,
        "colormode": "xy",
        "ct": 189,
        "effect": "none",
        "hue": 0,
        "on": false,
        "reachable": true,
        "sat": 254,
        "xy": [
            0.3012,
            0.1183
        ]
    },
    "swversion": null,
    "type": "Extended color light",
    "uniqueid": "84:2e:14:ff:fe:02:c0:3e-01"
}

ebaauw, 2024-04-10:
Ok, as I expected, the light doesn't expose _Color Capabilities_ (`colorcapabilities` is 0).  But the legacy code still exposes `state/xy`.  Can you change the light's colour through the API (PUT to `state` of `{"xy": [1.0, 0.0]}` should turn the light red-ish)?  Maybe this is a limitation of Phoscon?

Anyways, the best resolution is to create a proper DDF for the light, defaulting the colour capabilities.

LeoeLeoeL, 2024-04-10:
> Can you change the light's colour through the API (PUT to state of {"xy": [1.0, 0.0]} should turn the light red-ish)?

How can I do that?

LeoeLeoeL, 2024-04-10:
At this time, I think, it's beyond my possibilities.

Mimiix, 2024-04-10:
> At this time, I think, it's beyond my possibilities.

You can find out how to do it here https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

It's a very comprehensive guide

LeoeLeoeL, 2024-04-10:
Tks, I shall try.

LeoeLeoeL, 2024-05-02:
Keeping alive

Mimiix, 2024-05-02:
> Keeping alive

Did you test the DDF yet?

LeoeLeoeL, 2024-05-02:
Not yet.
In this time I tried to create DDF for Vimar Covering devices but with non luck and 3 days to have one device working again.

LeoeLeoeL, 2024-05-24:
Up

LeoeLeoeL, 2024-06-15:
Strange no other people suffer this issue

Mimiix, 2024-06-15:
Please test the DDF , otherwise I'm closing this.

LeoeLeoeL, 2024-06-15:
I'm not able to generate the DDF. If you can provide I'll be glad to test it.

Mimiix, 2024-06-15:
> I'm not able to generate the DDF. If you can provide I'll be glad to test it.

It's explained properly here 

https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

If you are not willing to follow or try at all, we are not able to get this resolved.

LeoeLeoeL, 2024-06-15:
Taking in account this issue is an effect of a previous deconz update, I think dresden-elektronik should provide to fix the regression.

Mimiix, 2024-06-15:
> Taking in account this issue is an effect of a previous deconz update, I think dresden-elektronik should provide to fix the regression.

And for that, some details are needed from the device.

As your not willing to provide the requested information, im closing this issue.

If you or anyone provides the info after closing, im happy to reopen.

closed 2024-06-15 by Mimiix

LeoeLeoeL, 2024-06-15:
Nice way to solve problems.
Instead asking me (I'm an user, not an expert), you should ask to the developers why these bulbs work OK in HUE, ZHA and Zigbee2MQTT and not in Deconz.

Mimiix, 2024-06-15:
> Nice way to solve problems. Instead asking me (I'm an user, not an expert), you should ask to the developers why these bulbs work OK in HUE, ZHA and Zigbee2MQTT and not in Deconz.

Because some user provided the required information :)

LeoeLeoeL, 2024-06-16:
if you like misteries, I have one just discovered.
If I switch on the lamp using TOOGLE command in deconz I see the colour.
If I switch on the lamp using ON command in deconz I see only white.

Regarding users that provide infos... I don't think HUE needs. Hue works perfectly with lamps always.
And, even if the device is not 100% zigbee compliant, it's a fact it works badly only in deconz.

TimSoethout, 2024-09-28:
It seems I have the same issue. I can however control color within the phoscon interface, but not in HA, there it only shows as an on/off switch.
This used to work correctly before last deconz add-on upgrade.

This could be related to https://github.com/home-assistant/core/pull/76305 for ZHA.

ChristophHornung, 2025-01-17:
Any update on this? I have the same issue, considering this issue is already a year old it looks like it wont be fixed?