dresden-elektronik/deconz-rest-plugin #7645: Zigbee Two-way Bilateral Energy Meter TZE200_rks0sgb7 - TS0601

gabriele-latino, 2024-02-27:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Zigbee Two-way Bilateral Energy Meter

### Manufacturer

_TZE200_rks0sgb7

### Model identifier

TS0601

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]
[image]
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

gabriele-latino, 2024-02-27:
[image]

gabriele-latino, 2024-03-20:
Hi, please add this device to deCONZ, thanks

gabriele-latino, 2024-04-05:
Is there any DDF I can try?

gabriele-latino, 2024-04-27:
Still needed......

Mimiix, 2024-04-27:
@Smanar can you help?

Smanar, 2024-04-27:
Hello, can try this DDF
```

{
    "schema": "devcap1.schema.json",
    "manufacturername": "_TZE200_rks0sgb7",
    "modelid": "TS0601",
    "product": "Two-way Bilateral Energy Meter",
    "sleeper": true,
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_CONSUMPTION_SENSOR",
            "restapi": "/sensors",
            "uuid": [
                "$address.ext",
                "0x01",
                "0x0702"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion",
                    "parse": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001",
                        "script": "tuya_swversion.js"
                    },
                    "read": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001"
                    }
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "config/offset",
                    "default": 0
                },
                {
                    "name": "config/on"
                },
                {
                    "name": "config/reachable"
                },
[... 267 lines cut ...]
                    },
                    "read": {
                        "fn": "none"
                    },
                    "default": 0
                },
                {
                    "name": "state/power",
                    "parse": {
                        "fn": "tuya",
                        "dpid": 111,
                        "eval": "Item.val = Attr.val;"
                    },
                    "read": {
                        "fn": "none"
                    },
                    "default": 0
                }
            ]
        }
    ]
}
```

Value will probably need adjustement (* 10 or / 10 for exemple), just give me the result and the value you want.

Smanar, 2024-05-08:
Good question, HassOS use too docker nope ?
What is your deconz version ? HA package, official docker image, custom installation ?
If you have access to the GUI, you can make a right clic on a node, edit DDF, then try to save it to see the folder used by deconz.

gabriele-latino, 2024-05-08:
Deconz 6.23.0 (Bump deCONZ to 2.25.3)
Installed ad HA's component, no custom settings

This is the "Save as" default location:

[image]

But I'm unable to acces this folder by SSH

Smanar, 2024-05-08:
Perhaps there is a docker for the addon ? Because it's the same folder than others users https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/20?u=smanar

For normal OS

> All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired.

closed 2024-06-07 by github-actions[bot]