dresden-elektronik/deconz-rest-plugin #7646: AVATTO ZWSM16-2-Zigbee

LeoeLeoeL, 2024-02-27:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

AVATTO ZWSM16-2-Zigbee

### Manufacturer

_TZ3000_mtnpt6ws

### Model identifier

TS0002

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering
[image]

### Electrical Measurement
[image]
[image]
[image]
[image]

### Any other cluster of relevance/interest

LeoeLeoeL, 2024-02-27:
[image]
If I set ON light 10 , light 11 switches to ON too.

SirBarkAlot, 2024-03-04:
i have one to, if there is support i would realy love it
if you would need somone to test it
PM me

LeoeLeoeL, 2024-03-05:
I have 4 of them.
2 are paired to zigbee2Mqtt and work OK.
[image]
I think the energy meter is disabled by firmware

LeoeLeoeL, 2024-04-17:
It could be a good thing to integrate this device because is not expensive.

Mimiix, 2024-04-17:
@Smanar Can you take a look if you have time :)?

Smanar, 2024-04-17:
Hello can you try this DDF 
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": [
    "_TZ3000_zmy4lslw",
    "_TZ3000_lmlsduws",
    "_TZ3000_mtnpt6ws",
    "_TZ3000_nuenzetq"
  ],
  "modelid": [
   "TS0002",
   "TS0002",
   "TS0002",
   "TS0002"
  ],
  "product": "Tuya 2 output module",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "refresh.interval": 86400,
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        },
        {
          "name": "config/tuya_unlock"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x02"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "refresh.interval": 86400,
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

Or if you want, just edit this DDF devices\tuya\_TZ3000_2channel_module.json

You need to re-include the DDF after that, because the tuya unlock sequence is done at inclusion or every 24h (you can wait 24h too)

LeoeLeoeL, 2024-04-17:
It has been impossible to load your DDF so I copied your items (violet, blue and green) and generated this one:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_mtnpt6ws",
  "modelid": "TS0002",
  "product": "TS0002",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/tuya_unlock",
          "public": false,
          "description": "Tuya sequence to unlock some device.",
          "refresh.interval": 86400,
          "default": false
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

Now the two ON/OFF switches are separated; If I switch on number 1 the number 2 stays off.
Too much easy. What's the catch?

Smanar, 2024-04-18:
So it mean the first DDF is working ?
Can add your device to the previous DDF officialy ?

Smanar, 2024-04-19:
>My DDF doesn't have 0x02 for the 2nd switch. Is it normal?

No ^^.
It's the endpoint, it's the number on title node, on your you have "01" "02 "F2", 0x01 is for the first gand and 0x02 for the second one.
So strange your DDF are working ....

BTW have added the device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7712

closed 2024-04-20 by manup