dresden-elektronik/deconz-rest-plugin #7647: CLIPSensors must be visible to the user? [improvement]

mattreim, 2024-02-27:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

These CLIPSensors must be visible to the user? They are actually only intended for internal processing and not for external processing.

Or can you get your own folder? (e.g. FakeSensors;)

### Steps to reproduce the behavior

Sensor list

### Expected behavior

No FakeSensors to see

### Screenshots

RESTED:

[image]

[image]

These have been added with version 2.26.1-beta.

ioBroker:

[image]

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

ebaauw, 2024-02-28:
The CLIP sensors should definitely be exposed over the API.  Whether they should be visible to users is up to the API client interacting with the user, ioBroker in your case?

ebaauw, 2024-02-28:
My external programs don't have this problem, as they explicitly offer the feature to interact with CLIP sensors.  It's up to the user to configure the program which sensors should be visible.  Phoscon as external program doesn't have this issue, as it explicitly filters out CLIP sensors.  Exposing CLIP sensors this way is according to spec, and not a bug.

mattreim, 2024-02-28:
Ok thanks, then I'll probably have to live with it.

mattreim, 2024-02-28:
Just to understand: the API cannot access other folders such as secrets?

[image]

ebaauw, 2024-02-28:
This looks like the database.  It contains tables, not folders.  And its structure is undocumented, and not exposed by the API.  See https://dresden-elektronik.github.io/deconz-rest-doc/ for an overview of the API.

Mimiix, 2024-02-28:
Closing as this is not a bug

closed 2024-02-28 by Mimiix