dresden-elektronik/deconz-rest-plugin #7653: Aqara Ceiling Light T1M (CL-L02D)

TUNER88, 2024-03-04:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Aqara Ceiling Light T1M

### Manufacturer

LUMI

### Model identifier

lumi.light.acn032

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]
[image]

### Level Control
[image]
[image]

### Color Control
[image]
[image]
=======
[image]
[image]
[image]

### Any other cluster of relevance/interest
[image]
[image]
[image]
[image]
[image]

pimw1, 2024-04-20:
This device has raving reviews all over the internet; from a commercial point of view it would (imho) be wise to bring support for it with all bells&whistles attached.

Mimiix, 2024-04-20:
If it was following the standard it should be working already out of the box.

@Smanar can you help with a DDF?

Smanar, 2024-04-20:
>If it was following the standard it should be working already out of the box.

Yes, exactly, what is the problem with this DDF ? What is working or not ?

ebaauw, 2024-05-17:
I would suspect the main lamp to work out of the box (as a _Color Temperature Light_; _Color Capabilities_ on the first endpoint correctly indicates it only supports _Color Temperature_.  The colour ring on the second endpoint should be exposed as an _Enhanced Color Light_.  The only issue with both endpoints is the ZHA device type: _Color Dimmable Light_; if memory serves, we needed whitelisting in the legacy code to correct that.  Creating a DDF for this would be pretty straightforward.  My main concern is that neither endpoint seems to support _PowerOn On/Off_.

The real challenge will be the effects on the colour ring.  I suspect they use the _LUMI_ cluster for these.  We would need some serious sniffing while the light is connected to the Aqara app to reverse engineer how to control these.  Probably need some serious JavaScript hacking for that, as we needed for the Hue gradient lights.

Paule-Panter, 2024-07-23:
Could anyone reopen this?

Mimiix, 2024-08-20:
What's the issue with the light? @Paule-Panter  please see the question of Smanar

Paule-Panter, 2024-08-20:
I can switch both endpoints on/off.
Both endpoint are exposed as "Color dimmable light", but the first (main) endpoint has physically only 2colors (cw and ww). At the moment i can't switch/fade between this colours. 
The outer ring has physically individual adressable LEDs.
It exposed as one "Color dimmable light", therefore it is only controlled as one light, so far.

ebaauw, 2024-08-20:
The device types are from ZHA.  In ZLL / ZB3 terms, the first endpoint (main light) is a Color Temperature Light; the second (aux light) an Enhanced Color Light.  Note that the second endpoint only contains RGB LEDs; colour temperature is emulated using these.

Both light endpoints seem to provide the same functions (apart, obviously, from the colour):
- _On/Off_ Cluster: Basic _On_, _Off_, and _Toggle_ commands.  No _Off with Effect_, no _On with Timed Off_.  No _Startup On/Off_ attribute.  Xiaomi have a manufacturer specific variant of this (0xFCC0/0x0517, which doesn't seem functional; it reports 0 and won't change when written;
- _Level Control_ cluster: basic commands, incl. _Move to Level (with On/Off)_ work.  _Options_, transition time attributes work.  _On Level_ just mirrors the current level and isn't settable.  No _Startup Current Level_.
- _Color Control_ cluster: first enpoint only `ct` (from 153 to 370), second endpoint `ct` (same range, but emulated using RGB LEDs) and `xy`.  No `hs`, no colorloop.  It doesn't report the gamut.  _Startup Color Temperature_ is non-functional.
- _Groups_ cluster: reports 8 groups for each endpoint.
- _Scenes_ cluster: still need to look into that, particularly if it supports `ct`.
- _Lumi specific_ cluster: as said before, _Poweron On/Off_ (0x0517) doesn't work.  Both endpoints report 10 segments (as per the attribute used for the LED strip).  The Aqara app supports (non-colour) effects on the main light as well.  These are using this cluster, but I need to look into the details.

Kinda bummer that it doesn't seem to support PowerOn settings, while the LED strip T1 does (see #7201).

ebaauw, 2024-08-20:
Created an initial DDF, but the `lights` resource for the first endpoint somehow exposes `state/xy`.  The corresponding attributes are also being polled.  I suspect this is the legacy code dealing with _Color Dimmable Light_.

Paule-Panter, 2024-08-20:
Thanks for this huge information and the ddf.
I'll try this after my holiday vacation. 

I would like to read the 'RTFM' for the Lumi specific cluster with the crazy states, where can i find this?

BR Paule

ebaauw, 2024-08-20:
I don't think there is a manual describing the LUMI specific cluster, at least not in a language I can read.

Most of our knowledge has been obtained through reverse engineering: changing stuff in the Aqara app while sniffing the Zigbee traffic between the Aqara hub and the device.  This knowledge is documented in GitHub issue, and summarised in `general.xml` (use the Source, Luke).

carpii, 2024-10-01:
Is this deviuce fully supported under DeConz now?

I'm thinking of buying one, but its unclear if there are still outstanding issues to resolve here

ebaauw, 2024-10-01:
No support yet for special scenes nor for gradients, see #7895.

oe-HH, 2024-10-04:
I'd like to use/test the respective DDF, however my Lamp seems not to be seen by my RaspBee II deCONZ-gui at all. Does somebody have any experience or link to a Howto with connecting the T1M with RaspBee family of ZigBee adapters? 
Thanks a lot!

ebaauw, 2024-10-04:
See the linked PR: standard light functionality for main and ring light should work, but special effects and gradients aren’t supported yet.

oe-HH, 2024-10-04:
> I'd like to use/test the respective DDF, however my Lamp seems not to be seen by my RaspBee II deCONZ-gui at all. Does somebody have any experience or a link to a HowTo-page for connecting the T1M with RaspBee family of ZigBee adapters? (Perhaps I have also just a hardware issue with my lamp ...) Thanks a lot!

I just figured out, that there was an issue with power supply caused by a defective wall switch which nin turn caused some undervoltage or so. So lamp was just "breathing" (probably indicating a undocumented fault state). 
After replacing the wall switch, the lamp behaved properly as described here and other forums using raspbee II, deCONZ and Home Assistant.

ebaauw, 2024-11-17:
Bump.

I finally managed to get the network key used by the Aqara M3 hub, by pairing an old device.  Still need to sniff the traffic between the hub and the ceiling light to reverse engineer the effects.

closed 2025-03-17 by github-actions[bot]

tayz98, 2025-04-06:
still not supported?

RasmusHo, 2025-07-25:
Still not fully supported?