dresden-elektronik/deconz-rest-plugin #7659: New Variant of MOES 1 GANG SWITCH TS0011 / _TZ3000

fifibest, 2024-03-09:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Moes Switch 1 Gang / MOES ZS-SR-EU1-WH-MS ZIGBEE 

### Manufacturer

_TZ3000_hhiodade

### Model identifier

TS0011

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

.

### Basic

[image]
[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

fifibest, 2024-03-09:
Actually, the DDF is available in "devices/tuya/_TZ3000_TS0011_1gang_switch_module.json", but it was missing this one specific model, adding "_TZ3000_hhiodade" to the list in "manufacturername" made the switch recognizable.

Can you add it to the code?

fifibest, 2024-03-10:
I have also noticed that with the MOES Ring switches I should be able to set the indicator light behaviour, but it is a read-only value right now. Can this be fixed?

[image]

It is also read-only for the 2-gang switch.

Can this be fixed?

Smanar, 2024-03-25:
You have tested if the DDF is working with your changes ?
Same for the second issue, pls test it first, IDK if the device is read only or not.
On file general.xml

```
        <attribute-set id="0x8001" description="Tuya special">
          <attribute id="0x8001" name="Back Light mode" type="enum8" default="0" required="o" access="r">
            <value name="Mode 1-Off" value="0"></value>
            <value name="Mode 2-Light when On" value="1"></value>
            <value name="Mode 3-Light when Off" value="2"></value>
            <value name="Mode 4-Actual state" value="3"></value>		  
          </attribute>
```

Just replace `access="r`" by `access="rw"` and restart deconz.

fifibest, 2024-03-25:
Yes, the DDF is working with my change, and I also tested setting the "Back Light mode" to rw, and set it to 3, and the indicator led on the switch behaves as it should, so please do both changes in deconz.

Thanks for the help!

Smanar, 2024-03-26:
PR Done, you have it here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7680