dresden-elektronik/deconz-rest-plugin #7660: Yagusmart Zigbee Smart Wall Light Switch 1 Gang No Neutral Wire Required

BottlecapDave, 2024-03-10:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Yagusmart Zigbee Smart Wall Light Switch 1 Gang No Neutral Wire Required

### Manufacturer

unknown

### Model identifier

unknown

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

[image]

[image]

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

This is a simple on/off light switch.

This used to work, but not sure when it stopped working.

If I go to Edit DDF, I get

[image]

I am accessing this via Home Assistant

Smanar, 2024-03-27:
Hello, try adding your device in this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0011_1gang_switch_module.json

Remember you need to add Manufacture name AND model id.

BottlecapDave, 2024-04-07:
I edited this manually in the deconz view within HA and I managed to communicate with the device again. 

I've opened https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7699 to add the profile

Smanar, 2024-04-07:
Thx, but why you haven't just added your device to the existing list ?
There new tuya clone every day, so we try to "group" them.

BottlecapDave, 2024-04-15:
I was trying to follow the convention laid out by existing files (quite confusing of when you add to one vs when you create a new one). The file copied is just what's been working in my instance so I know it works. If the owners of this repo would like me to update my PR and just update one of the existing files, I'll do that.

Smanar, 2024-04-15:
>quite confusing of when you add to one vs when you create a new one

Lol, right, I can understand, no logic.

>The file copied is just what's been working in my instance so I know it works

Yep, but if you can make a try with the bind/report working mode ? if it work, will be better for your network, the poll make lot of more request than the bind/report mode.

closed 2024-05-13 by manup