dresden-elektronik/deconz-rest-plugin #7661: Bosch Light/shutter control unit II

sahnetorti, 2024-03-10:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Light/shutter control unit II

### Manufacturer

Bosch

### Model identifier

RBSH-MMS-ZB-EU

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Simple Metering
[image]

### Electrical Measurement
[image]
[image]
[image]
[image]

### Window covering
[image]
[image]
[image]

###Diagnostics
[image]

This device is a bit special as it has a "split personality": it can function as 2x switch (i.e. for controlling an on/off light) OR as a window covering device. For the latter it must be configured, which can be done  using the Bosch Gateway & proprietary app.

The magic seems to be in the cluster #FCA0, where some attributes must be set accordingly. Unfortunately these cluster attributes are not available in deconz. A reference to these attributes can be found in a discussion for ZHA [here](https://github.com/zigpy/zha-device-handlers/issues/2518)

BTW I was able to pair this device through the "pair with install code" method in the REST API.

Smanar, 2024-03-27:
To add the feature in deconz, edit the general.xml file and for

```
service: zha_toolkit.attr_write
data:
  ieee: ...
  manf: 0x1209
  endpoint: 1
  cluster: 0xfca0
  attribute: 0x0000
  attr_type: 0x30
  attr_val: 0x01
```

The result will be

```
   <cluster id="0xfca0" name="Bosch specific" mfcode="0x1209">
        <description>Bosch Specific cluster.</description>
        <server>
        <attribute id="0x0000" type="enum8" name="shutter mode" access="rw" required="o" mfcode="0x1209">
            <value name="xxxx" value="0x00"></value>
            <value name="yyyyy" value="0x01"></value>
        </attribute>
        <attribute id="0x0002" type="u32" name="calibration closing time" access="rw" required="o" mfcode="0x1209"></attribute>
        </server>
        <client>
        </client>
    </cluster>
```

So you miss feature, but the device is correclty reconized by defaut ? Or you need a DDF too ?

sahnetorti, 2024-03-31:
Hi smanar,

  thanks for providing the relevant XML snippets. I added this to the general.xml:
```
    <cluster id="0xfca0" name="Bosch specific" mfcode="0x1209">
      <description>Bosch Specific cluster.</description>
      <server>
        <attribute id="0x0000" type="enum8" name="shutter mode" access="rw" required="o" mfcode="0x1209">
          <value name="Switch mode" value="0x00"></value>
          <value name="Rolershutter mode" value="0x01"></value>
        </attribute>
        <attribute id="0x0002" type="u32" name="calibration closing time" access="rw" required="o" mfcode="0x1209">
        </attribute>
      </server>
      <client>
      </client>
    </cluster>
```
which made the shutter work. I am not sure with the names for the modes&values, though, it might be that this is not accurate. I set the device to "rolershutter mode" / 0x01 and after that I could control the shutter (up/down/stop).

Additionjally I had to set the "calibration closing time" in my setup to make the shutters extend/pull back the full way. Without it they extended/pulled back only about 80%.

Also, reading the manual more thoroughly it mentions that when the device is "unconfigured" (no mode set) the inputs S1/S2 for physical switches will not work. I cannot confirm this as I do not have or need physical switches in my setup.

I will be happy to test the changes in the PR mentioned above when I find the time. Is there a better way to do it except adding the new sections manually to my local files?

Smanar, 2024-04-01:
@SwoopX have started a PR for your device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7683
You have a DDF and a more complete code to edit for the xml

sahnetorti, 2024-04-03:
I have applied the changes manually to my installation. I can confirm that
- "Device mode"
- "Calibration opening time"
- "Calibration closing time"
- "Motor state"
all work.

I can control the shutters through Phoscon and through Homeassistant (via deconz integration).

In my install I cannot test the settings for
- "Child lock"
- Switch type"

PR #7683 (2024-04-03): DDF for Bosch Light/shutter control II
Interesting device. Making an educated guess here, especially on the switch resource and the button map. Would be great if it could be tested upfront.

Closes #7661
closed 2024-04-03 by manup