dresden-elektronik/deconz-rest-plugin #7663: Ikea Badring

Emlben, 2024-03-10:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Ikea Badring

### Manufacturer

Ikea

### Model identifier

BADRING Water Leakage Sensor

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest 
(My first attempt to provide this information - please bare with me ;) )
[image]

Hector47, 2024-04-02:
I also have the same sensor. How can I help ? Do you have an DDF a can use as base ?

danielchansen, 2024-04-02:
Same. Have four of these sensors. Anything I can do to contribute?

Mimiix, 2024-04-03:
@Smanar can you take a look?

Smanar, 2024-04-03:
Someone have started a DDF here https://forum.phoscon.de/t/ddf-for-ikea-device-badring-water-leakage-senor/4736

RP-Develop, 2024-04-07:
This DDF don't run. I get no alarm from the sensor. Only if I read it manually, so I get a changing in 0x0500 / 0x0002.
[image]
[image]
Is it possible to set the sensor so that it sends the alarm?

Smanar, 2024-04-07:
Hello can you try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_IKEA",
  "modelid": "BADRING Water Leakage Sensor",
  "vendor": "IKEA",
  "product": "BADRING Water Leakage Sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WATER_LEAK_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/productid",
          "refresh.interval": 86400
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/alert",
          "default": "none"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 86400,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Math.round(Attr.val / 2)",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/group"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "config/enrolled",
          "public": true
        },
        {
          "name": "state/water",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

You miss the enrollement on previous one.

RP-Develop, 2024-04-07:
Hi Smanar,

sorry, but it don't work. I don't get the alarm. The state has no values. 

some from the Phoscon Web:
[image]
[image]

Smanar, 2024-04-07:
`"enrollled" : null.`

The enrollement have failed. Have you tried to re-include it ?

RP-Develop, 2024-04-07:
maybe helpful:

When I read out the IAS manually, I get these messages in the debug view of the IAS::

19:13:26:010 Force binding of attribute reporting for node BADRING Water Leakage Sensor 2
19:13:56:091 [IAS ZONE] - Address 0xB0C7DEFFFE67151B, Payload 0000003000010000312a00020000190000100000f00000000000000000, Command 0x01
19:13:56:093 [IAS ZONE] - 0xB0C7DEFFFE67151B No IAS sensor found for endpoint: 0x01
19:13:57:100 [IAS ZONE] - Address 0xB0C7DEFFFE67151B, Payload 11000020ff, Command 0x01
19:13:57:101 [IAS ZONE] - 0xB0C7DEFFFE67151B No IAS sensor found for endpoint: 0x01

RP-Develop, 2024-04-07:
> `"enrolled" : null.`
> 
> The enrollement have failed. Have you tried to re-include it ?

yes:
[image]

I started with your DDF.

Smanar, 2024-04-07:
Ha my bad, I forget the fingerprint

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_IKEA",
  "modelid": "BADRING Water Leakage Sensor",
  "vendor": "IKEA",
  "product": "BADRING Water Leakage Sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WATER_LEAK_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "attr/productid",
          "refresh.interval": 86400
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/alert",
          "default": "none"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 86400,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Math.round(Attr.val / 2)",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/group"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "config/enrolled",
          "public": true
        },
        {
          "name": "state/water",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

RP-Develop, 2024-04-07:
[image]
[image]

RP-Develop, 2024-04-07:
[image]

RP-Develop, 2024-04-07:
[image]

Running on MAC mini M2 Sonoma 14.4.1

SwoopX, 2024-04-07:
The device presumably needs to be poked on the IAS Zone cluster to start the enrollment process, like the Parasoll (and the battery binding was missing). This one should do the trick:

```{
  "schema": "devcap1.schema.json",
  "manufacturername": "$MF_IKEA",
  "modelid": "BADRING Water Leakage Sensor",
  "vendor": "IKEA",
  "product": "BADRING Water Leakage Sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WATER_LEAK_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/productid",
          "refresh.interval": 86400
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 86400,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Math.round(Attr.val / 2)",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "config/enrolled",
          "public": true
        },
        {
          "name": "state/water",
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1,alarm2"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0500",
            "at": "0x0000"
          },
          "awake": true,
          "refresh.interval": 84600
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 43200,
          "change": "0x00000002"
        }
      ]
    }
  ]
}
```

RP-Develop, 2024-04-08:
I have also tried this DDF, but unfortunately without success. I have also used the complete DDF from Parsoll Sensor, adapted accordingly of course, but unfortunately without success. I can't get a status change for "state/water". Only when I use "read" in 0x0500 in deConz do I see a status change in at 0x0002 and also an event in Phoscon. 

After the "read" is also the at 0x0000 changed to Enrolled.
[image]

Can I log some more? 

The IAS log in moment when I let beep the sensor and use the read button:

21:03:18:977 [IAS ZONE] - Address 0xB0C7DEFFFE62A297, Payload 0000003001010000312a00020000190100100000f0ff4305ffff2e2100, Command 0x01
21:03:18:978 [IAS ZONE] - 0xB0C7DEFFFE62A297 Read attributes response:
21:03:18:979 [IAS ZONE] - 0xB0C7DEFFFE62A297   -> IAS Zone State: enrolled.
21:03:18:980 [IAS ZONE] - 0xB0C7DEFFFE62A297   -> IAS CIE address = 0x00212EFFFF0543FF: already written.
21:03:18:980 [IAS ZONE] - 0xB0C7DEFFFE62A297 Sensor (ZHAWater) is enrolled.
21:03:19:931 [IAS ZONE] - Address 0xB0C7DEFFFE62A297, Payload 1100002064, Command 0x01
21:03:19:932 [IAS ZONE] - 0xB0C7DEFFFE62A297 Read attributes response:
21:03:19:933 [IAS ZONE] - 0xB0C7DEFFFE62A297 Sensor (ZHAWater) is enrolled.

...and the same without beep:

21:05:26:913 [IAS ZONE] - Address 0xB0C7DEFFFE62A297, Payload 0000003001010000312a00020000190000100000f0ff4305ffff2e2100, Command 0x01
21:05:26:914 [IAS ZONE] - 0xB0C7DEFFFE62A297 Read attributes response:
21:05:26:915 [IAS ZONE] - 0xB0C7DEFFFE62A297   -> IAS Zone State: enrolled.
21:05:26:916 [IAS ZONE] - 0xB0C7DEFFFE62A297   -> IAS CIE address = 0x00212EFFFF0543FF: already written.
21:05:26:916 [IAS ZONE] - 0xB0C7DEFFFE62A297 Sensor (ZHAWater) is enrolled.
21:05:27:885 [IAS ZONE] - Address 0xB0C7DEFFFE62A297, Payload 1100002064, Command 0x01
21:05:27:887 [IAS ZONE] - 0xB0C7DEFFFE62A297 Read attributes response:
21:05:27:887 [IAS ZONE] - 0xB0C7DEFFFE62A297 Sensor (ZHAWater) is enrolled.
21:05:27:933 [IAS ZONE] - Address 0xB0C7DEFFFE62A297, Payload 1100002064, Command 0x01
21:05:27:934 [IAS ZONE] - 0xB0C7DEFFFE62A297 Read attributes response:
21:05:27:934 [IAS ZONE] - 0xB0C7DEFFFE62A297 Sensor (ZHAWater) is enrolled.

many thanks

Smanar, 2024-04-09:
Nothing to see but why this part on both DDF (this one and the praraoll)
```

          "read": {
            "fn": "zcl:attr",
            "ep": 2,
            "cl": "0x0500",
            "at": "0x0000"
          },
```

There is no endpoint 0x02, and we are not using this attribute (the 0x0000)
Not a problem as we don't use poll, but useless for me.

For your issue, all seem fine for me, the enrollement is done, you haven't log if the device detect water, without using the "read" button ?

SwoopX, 2024-04-09:
> and we are not using this attribute (the 0x0000)
> Not a problem as we don't use poll, but useless for me.

In fact, attribute 0x0000 is imperative for any IAS Zone device correctly implementing the standard and deconz always uses it to determine the current state and if there's any to do.

I, personally, would put my bet on a missed device reset on every DDF update. This would explain everything what has been described here and also why things didn't work out as expected.

RP-Develop, 2024-04-09:
> For your issue, all seem fine for me, the enrollement is done, you haven't log if the device detect water, without using the "read" button ?

No, nothing. I have two devices installed at the moment. I have the feeling that the battery event is also not like from other devices.

RP-Develop, 2024-04-09:
> would put my bet on a missed device reset on every DDF update

Can I influence this?

openKimchi, 2024-04-10:
I installed the sensor with this DDF. Then restarted the entire Deconz server. When triggered manually, the alarm jumps to true (state in ioBroker) and stays that way. When triggered again manually, it jumps to false and stays that way. And the battery indicator doesn't work at all.

RP-Develop, 2024-04-10:
It looks as if the sensor is going into sleep mode. So if you have it read manually, do not wake it up. The first reaction after waking up is to hit the sensor with water and hear the beep. You can then see in the IAS log that all read commands have been processed.

openKimchi, 2024-04-10:
By manual triggering I mean wetting it with water...almost embarrassing but where can I find this IAS log?

RP-Develop, 2024-04-10:
(macOS) under Help - Debug view - select IAS
[image]

openKimchi, 2024-04-10:
ok, thx. Got;

`14:23:39:026 [IAS ZONE] - Address 0x38398FFFFE0AE1FD, Payload 000000640000, Command 0x00
14:23:39:027 [IAS ZONE] - 0x38398FFFFE0AE1FD Status Change, status: 0x0000, zoneId: 100, delay: 0
14:23:39:027 [IAS ZONE] - 0x38398FFFFE0AE1FD Sensor (ZHAWater) is enrolled.`

RP-Develop, 2024-04-10:
> 14:23:39:026 [IAS ZONE] - Address 0x38398FFFFE0AE1FD, Payload 000000640000, Command 0x00 
14:23:39:027 [IAS ZONE] - 0x38398FFFFE0AE1FD Status Change, status: 0x0000, zoneId: 100, delay: 0 
14:23:39:027 [IAS ZONE] - 0x38398FFFFE0AE1FD Sensor (ZHAWater) is enrolled. 

I don't get it after wetting.

On which system is running your deConz?

openKimchi, 2024-04-10:
> On which system is running your deConz?

it's a Ubuntu 22.04.4 LTS Server as a VM on a Proxmox System.

openKimchi, 2024-04-10:
> One question: Why do I get a different view between the preview of the DDF editor and the file that I have edited with an external editor?

Same for me

Smanar, 2024-04-10:
>One question: Why do I get a different view between the preview of the DDF editor and the file that I have edited with an external editor?

Honnestly there is so much "strange" thing that happen in the editor, don't take care of it. Just look at the file name on the window title and remember deconz check DDF only at start.

Mimiix, 2024-04-10:
> One question: Why do I get a different view between the preview of the DDF editor and the file that I have edited with an external editor?
> 
> <img alt="image" width="1286" src="https://private-user-images.githubusercontent.com/156454090/321249273-33deb3b0-8836-4447-8fe9-53d07beaa04e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI3NzMwNzksIm5iZiI6MTcxMjc3Mjc3OSwicGF0aCI6Ii8xNTY0NTQwOTAvMzIxMjQ5MjczLTMzZGViM2IwLTg4MzYtNDQ0Ny04ZmU5LTUzZDA3YmVhYTA0ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQxMFQxODEyNTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iODQ5YWZlMWZlMWMzNGMxYmJmNzk4NGZjNTNjMGEwNjlmNDg0ZjUwMjZhMWZiNzBhZjgyOTA5OGIxNWQ2Mzc1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.bGknWGO2wEwSgRtnzkLUt3XeJypxiisKE4amE2kxnXc">

It probably didn't load or the file is in the wrong location.

RP-Develop, 2024-04-10:
the file should be loaded correctly, because I see the file name at the top line of editor frame.
[image]

The real path on my MAC is: /Users/home/Library/Application Support/dresden-elektronik/deCONZ/devices/badring_water_leakage_sensor.json

RP-Develop, 2024-04-12:
I haven't touched anything for more than 24 hours now. I didn't get anything back. No battery status and no alarm - nothing

openKimchi, 2024-04-12:
I just tried it again and everything still works. So it can't be the DDF file. But it only worked after I completely restarted the server. This is done quickly with Proxmox.

RP-Develop, 2024-04-12:
Can you tell me the version of deConz? My version is v2.25.3.

openKimchi, 2024-04-12:
2.26.3

openKimchi, 2024-04-12:
2.26.3, the latest

RP-Develop, 2024-04-12:
2.26.x don't work on my MAC - [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7698]

Smanar, 2024-04-12:
> ok, thx. After wetting it I got;

`14:23:39:027 [IAS ZONE] - 0x38398FFFFE0AE1FD Status Change, status: 0x0000, zoneId: 100, delay: 0`

So your device react to water, but you need to have other value than 0x0000 for status

DanielHabenicht, 2024-04-13:
The DDF also does not work for me (on homeassistant with deCONZ 2.25.3): 

IAS Log: 
[image]

Nothing shows up in HA either, batter at 0 at no events registered.

Smanar, 2024-04-14:
But on your logs all is fine, we can see the status go from 0x0000 to 0x0001.
Can you make the same log but with the flag "DDF" too ?

DanielHabenicht, 2024-04-14:
I looked it up today and it now reports the battery (8 hours after initially enrolling it). But I do not receive any events in Homeassistant (is that a part that still must be programmed in the Homeassistant add on or should this work out of the box?)

Smanar, 2024-04-14:
>is that a part that still must be programmed in the Homeassistant add on or should this work out of the box

For me it's out of the box, can check direclty in the API to be sure, phoscon/help/API Information/sensors.

DanielHabenicht, 2024-04-14:
ok, yes whatever the system had... It's working now. Maybe it just had to do with all the manual stuff involved to add the DDF. 
Works now: 
[image]

poelzi, 2024-04-18:
I also had quite some trouble getting this sensor to work. After deleteing and pairing again with the latest DDF and triggering the sensor, everything works.

closed 2024-04-20 by manup