dresden-elektronik/deconz-rest-plugin #7667: Aqara C1 Pet Feeder PETC1-M01

mcormick, 2024-03-15:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Aqara Smart Pet Feeder C1

### Manufacturer

Aqara

### Model identifier

aqara.feeder.acn001

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
N/A
### On/Off
[image]
### Level Control
N/A
### Color Control
N/A
### Thermostat
N/A
### Simple Metering
N/A
### Electrical Measurement
N/A
### Groups
[image]
### Identify
[image]
### Lumi Specific
[image]
[image]
[image]
### Otau
[image]
### Scenes
[image]
### Time
[image]

mcormick, 2024-03-15:
https://forum.phoscon.de/t/aqara-pet-feeder-c1/4740

SwoopX, 2024-03-15:
Duplicate of #6595, therefore closing

closed 2024-03-15 by SwoopX

mcormick, 2024-03-15:
Rucksichtlos closing a new request because a 1.5 year old one was already closed, without satisfying result I might add? Surely this could be looked into again with all the work that has been done in deCONZ in the months thereafter? Also, the new cluster screenshots differ to a certain degree from this original request. Politely requesting a reopening of either this, or the old request and a serious look into possible support for this device.

francescopeloi, 2024-03-18:
@SwoopX kindly tagging you to see if you agree with the above message?

SwoopX, 2024-03-20:
Apologies, but I do not understand what is reckless in this regard. We have rules here and give guidance on what to do prior to raising new device requests. This is available on the one hand to point the potential requestor at what to do before putting work into creation of an issue (which admittedly can be rather extensive depending on the device), and on the other hand to not unnecessarily blow up the number of device issues per se and/or scatter any device relevant information over various issues. Specifically the latter is very inconvenient and error prone and this just makes the lives of the developers harder as it needs to be.

It is unfortunate that the result of the already available issue is unsatisfying for you, but this is not an acceptable justification. You're always free to request re-opening issues and welcome to contribute with newly available or missing information to resume work. Please do note though that often information about functionality, issues or blockers is already documented in the respective issues.

Thanks for understanding!