dresden-elektronik/deconz-rest-plugin #7671: deCONZ.app on macOS missing native ARM support

apolgar-ms, 2024-03-16:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

The deCONZ build for macOS is currently only available for the x86_64 architecture. While this build _does_ run on Apple's ARM chips (M1, M2, M3, etc) through the Rosetta compatibility layer, Apple no longer sells Intel-based Macs, and new OS releases for Intel chips will be discontinued after this year.

### Steps to reproduce the behavior

N/A

### Expected behavior

In addition to the existing x86_64 deCONZ build for macOS, there should be a build for the ARM-based Apple Silicon architecture. Ideally, this should be a Universal app, combining both binaries in a single application bundle.

The chip architecture transition was announced back in 2020.

### Environment

Apple Silicon-based Macs

### Additional context

https://developer.apple.com/documentation/apple-silicon/building-a-universal-macos-binary

https://www.apple.com/newsroom/2020/06/apple-announces-mac-transition-to-apple-silicon/

manup, 2025-05-02:
Done, the native Apple Silicon builds are now included since [v2.29.3-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.29.3-beta)

closed 2025-05-02 by manup

porelli, 2025-11-08:
Is it intentional that recent releases only provide ARM builds for macOS?

Reference: https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.32.0-beta