dresden-elektronik/deconz-rest-plugin #7682: I have an issue with model is TS0601 - TZE200_myd45weu reporting the wrong value

alphatech69, 2024-03-27:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Steps to reproduce the behavior

The sensor is reporting the wrong info, this is a soil moisture device

### Expected behavior

The sensor is reporting the wrong info, this is a soil moisture device

### Screenshots

I have attached the screenshots

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

### deCONZ Logs

PC, Homeseer and Home Assistant with decxonz.

Mimiix, 2024-03-27:
Hi, not sure what the issue is. Also, the template is not complete.

Smanar, 2024-03-27:
You have the issue only on phoscon ?
It's fine on HA ?

You can too see the device direclty in the API, on phoscon/help/API Information/sensor

alphatech69, 2024-03-27:
> Hello, you have the issue only on phoscon ?

Thanks for your help. I have attached the screenshots that shows what I see
in all the apps, not sure if you are seeing the same thing that I see but
it shows that the sensor in ha report moisture to 0, that is incorrect.  On
the phoscon it shows the device as thermostat that is incorrect and in
homeseer it only register the temperature sensor.  As I mentioned before,
this is a soil device it should register the soil moisture.

Mimiix, 2024-03-27:
> Thanks for your help. I have attached the screenshots that shows what I see in all the apps, not sure if you are seeing the same thing that I see but it shows that the sensor in ha report moisture [...]

Is it in the ground?

alphatech69, 2024-03-27:
It is in a plant right now.

Mimiix, 2024-03-27:
Looks like this device has bad firmware, or at least a high interval by reading this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6731

As this is not a bug report as described in #5113 I am closing this issue.

closed 2024-03-27 by Mimiix