dresden-elektronik/deconz-rest-plugin #7684: Device Request: TY0201 _TZ3000_bjawzodf

4MeInfo, 2024-03-28:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

TY0201

### Manufacturer

Tuya

### Model identifier

_TZ3000_bjawzodf

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]
<img width="1051" alt="Bildschirmfoto 2024-03-28 um 21 
[image]
25 46" src="https://github.com
<img width="1051" alt="Bildschirmfoto 2024-03-28 um 21 25 13" src="https://github.com/dresden-ele
[image]
ktronik/deconz-rest-plugin/assets/53448222/3374a
[image]
bcd-5074-434f-9563-95d7daac512c">
/dresden-elektronik/deconz-rest-plugin/assets/53448222/af7bd224-799c-47a0-b114-7c80c22060cd">

### Basic

[image]
[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

4MeInfo, 2024-03-28:
https://de.aliexpress.com/item/1005005454107609.html?spm=a2g0o.order_list.order_list_main.6.12fe5c5fyoZDaC&gatewayAdapt=glo2deu

4MeInfo, 2024-03-28:
https://de.aliexpress.com/item/1005005951863680.html?spm=a2g0o.order_detail.order_detail_item.19.571d6368Hbd0hu&gatewayAdapt=glo2deu

4MeInfo, 2024-03-29:
[_TZ3000_bjawzodf_temp_hum_sensor.json](https://github.com/dresden-elektronik/deconz-rest-plugin/files/14800970/_TZ3000_bjawzodf_temp_hum_sensor.json)

I modify some existing file and it works.
[image]

4MeInfo, 2024-03-29:
[_TZ3000_bjawzodf.json](https://github.com/dresden-elektronik/deconz-rest-plugin/files/14801022/_TZ3000_bjawzodf.json)
[image]

I modify some existing file and it works.

closed 2024-03-29 by 4MeInfo

Smanar, 2024-03-29:
Ha yes there is already a DDF, but not for the same model

```
  "manufacturername": "_TZE200_bjawzodf",
  "modelid": "TS0601",
```

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_bjawzodf_humidity_temp.json

If it's the one you have modified, can you submit a PR ?

4MeInfo, 2024-03-29:
I used the [https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0201_temp_hum_sensor.json](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0201_temp_hum_sensor.json)

Smanar, 2024-03-29:
Ha yes, one DDF work with tuya cluster, the second one with classics clusters.
And your device have both ....

Lol, so impossible to guess without the device.

TheNON75, 2024-10-11:
hi @SwoopX, 

what is the progress on this?
Just bought one myself too. Can I support you somehow?

TheNON75, 2024-10-11:
Ok, I will try it later tonight

TheNON75, 2024-10-11:
it is kind of weird stuff.
phoscon state: not reachable, measurements from yesterday are shown
deconz gui: temp and rh clusters shows recent data
REST API: same as phoscon

Let me know what information you need, I will try to provide them, however I will be busy during the weekend by running around

Smanar, 2024-10-12:
Yep, the battery status can need some time.
Tell me if it works to make the PR, or you can make it too if yoy want, just need to add the model to this DDf https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0201_temp_hum_sensor.json

TheNON75, 2024-10-12:
Well, battery and version are still not okay, but apart from this the device works.

Do you want to dig into it, or see it later if it can be fixed at all? If the latter, then please proceed with a pr.

Smanar, 2024-10-13:
Can try to fix the batrtery first, perhaps this device need something not standard.
Can you check the cluster 0x0001, to see if the attributes 0x0021 is used ?

On the DDF, can try replace 

```
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
```

by

```
        {
            "name":"config/battery",
            "refresh.interval": 86400,
            "read":{
                "at":"0x0021",
                "cl":"0x0001",
                "ep":1,
                "fn":"zcl:attr"
            },
            "parse":{
                "at":"0x0021",
                "cl":"0x0001",
                "ep":1,
                "eval":"Item.val = Attr.val/2"
            },
            "default":0
        },
```

TheNON75, 2024-10-13:
The adjusted part is in place, waiting for updated data.

TheNON75, 2024-10-14:
@Smanar after 24H the battery is still not updated. Fyi in the gui it show simply 76
[image]

[image]

Smanar, 2024-10-15:
76 ?
Your device have a dead battery, or this device don't need the / 2.
You haven't another battery to make try ?

Another thing, when you read a value in the GUI, it update the value in the API too, so you need to have a 76/2 value for battery now.

Else can you enable log in deconz with only flag "DDF", will be not too much talkative and try to read this value again ?

TheNON75, 2024-10-17:
Battery is swapped with an Ikea vallhorn where the battery was reported to be 200.
The batteries from the tuya reported in Ikea as 166
The batteries from Ikea reported in Tuya as 192 (instead of 200, see above), but .... (!!!!!)
Apparently this value was reported after insertion. By the next reporting it became c0 (!!!), which is in fact 192....

[image]

Smanar, 2024-10-17:
So fisrt bug, it seem sometime deconz show hexa value instead of decimal.

So the battery level is good on the API, but not on Phoscon ?
This device create 2 entries, a ZHATEmperature and a ZHAHumidity, both have good battery level ? Perhaps Phoscon use the bad one ?

TheNON75, 2024-10-17:
I went back to 2.22 recently as aqara sensors were reporting very sporadically, sometimes for 4-6hours no updates at all.

Correct, good in API but not in phoscon.
No, humidity battery level is 0% (parsing is missing) and I agree, phoscon picks the second one (0%)
Imo battery shall be kept only for one sensor or perhaps add zhabattery(?)

TheNON75, 2024-10-18:
On 2.28.1 the battery is shown in phoscon properly (however this version is less good for aquaria sensors)
The device is currently not reachable but it is normal as decent was restarted recently due to switching to latest stable.

Only the version is not defined yet in ddf and so it is missing
[image]

Smanar, 2024-10-18:
It's definied on the DDF

```
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
```

You have put the DDF on the "tuya" folder ? there is a relative path.

>No, humidity battery level is 0% (parsing is missing) and I agree, phoscon picks the second one (0%)
Imo battery shall be kept only for one sensor or perhaps add zhabattery(?)...in the other hand, the same battery level can be reported to both sensors if this is the standard

So it work for an entry and not the other ?
I m checking the DDF again, and I m seing the same code

```
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
```

Or you are using the modified code only on one of them ?

```
        {
            "name":"config/battery",
            "refresh.interval": 86400,
            "read":{
                "at":"0x0021",
                "cl":"0x0001",
                "ep":1,
                "fn":"zcl:attr"
            },
            "parse":{
                "at":"0x0021",
                "cl":"0x0001",
                "ep":1,
                "eval":"Item.val = Attr.val/2"
            },
            "default":0
        },
```

TheNON75, 2024-10-24:
You're right, adjusted only one of the batteries (however in the latest stable it is picked and shown properly)

The file is in the ```
/opt/deCONZ/devices``` (I am on docker) folder and this explains why the js did not run. Do you want me to copy it over or we accept the theory "it should work"?

Smanar, 2024-10-24:
>or we accept the theory "it should work"

Lol, yes I think we can, it's the same code on ALL tuya DDF.

So do you want to submit a PR or want I make it ?

TheNON75, 2024-10-24:
Let's accept it should work :) If you don't mind, I would leave the pr for you to avoid that I screw up something accidentally

Smanar, 2024-10-25:
Done > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7993