dresden-elektronik/deconz-rest-plugin #7686: Deconz: Unexpected light effect type in Home Assistant

pimw1, 2024-03-30:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Since upgrading from deCONZ from v2.25.3 -> v2.26.3, it is throwing the following errors in Home Assistant:

Logger: pydeconz.models.light.light
Source: components/deconz/light.py:203
First occurred: 10:38:07 (8 occurrences)
Last logged: 10:38:07

Unexpected light effect type 0x5ec4
Unexpected light effect type 0xc2c
Unexpected light effect type 0x6041
Unexpected light effect type 0x5a55

I've initially created an issue in the Home Assistant repo: https://github.com/home-assistant/core/issues/114430. [Kane610](https://github.com/Kane610) reached out to the deCONZ developers and was given the advice that i should create an issue here in the deCONZ repo. My initial thought was that it was related to the new beta version of Home Assistant (2024.4.0b3), but that assumption turned out to be incorrect (i installed the home assistant beta and the new deconz version at the very same day).

Therefore, i am putting the same information here:

I've created an diagnostic file and added it to this issue. I've already done analysis on this file by checking which unique product id and product names are related to which effect type. From this analysis, the following can be concluded:

0x5ec4 -> 'productid': 'Philips-LCA006-1-A60HECLv1', 'productname': 'Hue color lamp' (configured 'name': 'Gang Lamp Plafond')
0x5a55 -> 'productid': 'Philips-LCA006-1-A60HECLv1', 'productname': 'Hue color lamp' (configured 'name': 'Gang Lamp Staand')
0xc2c -> 'productid': 'Philips-LCE002-1-E14ECLv2', 'productname': 'Hue color candle' (configured 'name': 'Keuken Lamp Keukenblok 3')
0x6041 -> 'productid': 'Philips-LCA001-5-A19ECLv6', 'productname': 'Hue color lamp' (configured 'name': 'Slaapkamer Lamp Lantaarn')

I've related these specific lights to the gui of deconz, by making screenshots of the Hue Effects cluster and Hue Entertainment cluster of the respective lights:

0x5ec4 -> 'productid': 'Philips-LCA006-1-A60HECLv1', 'productname': 'Hue color lamp', 'name': 'Gang Lamp Plafond':
[image]
[image]

0x5a55 -> 'productid': 'Philips-LCA006-1-A60HECLv1', 'productname': 'Hue color lamp', 'name': 'Gang Lamp Staand'
[image]
[image]

0xc2c -> 'productid': 'Philips-LCE002-1-E14ECLv2', 'productname': 'Hue color candle', 'name': 'Keuken Lamp Keukenblok 3':
[image]
[image]

0x6041 -> 'productid': 'Philips-LCA001-5-A19ECLv6', 'productname': 'Hue color lamp', 'name': 'Slaapkamer Lamp Lantaarn'
[image]
[image]

### Steps to reproduce the behavior

1. Install deconz v.2.26.3
2. make sure you have connected the mentiond light bulbs from Philips Hue to your zigbee deCONZ network
3. simply reboot home assistant 
4. check the log files of home assistant

### Expected behavior

no errors should be visible

### Screenshots

see above

### Environment

- Host system: 
pc Intel Core i5
- Running method:
Proxmox -> VM, deCONZ running in docker
Proxmox -another VM, Home Assistant OS running inside
- Firmware version: 26780700 (Conbee II)
- deCONZ version (not Home assistant Addon version!): v2.26.3
- Device:  ConBee II
- Do you use an USB extension cable: yes (although not relevant for this issue)
- Is there any other USB or serial devices connected to the host system? Yes, HDD (although not relevant for this issue)

### deCONZ Logs

[home-assistant_deconz_2024-03-29T09-38-37.138Z.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/14811405/home-assistant_deconz_2024-03-29T09-38-37.138Z.log)

pimw1, 2024-04-21:
The bug is relevant and pausing the upgrade of the Home Assistant addon.

yan14, 2024-04-21:
Why is nobody taken care of it?

ebaauw, 2024-04-21:
I think this has been addressed in https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7704

pimw1, 2024-04-21:
Great, thanks! I'll test with the next beta release.

pimw1, 2024-05-13:
I'll keep this open and report back when the next beta is published.

pimw1, 2024-05-15:
The issue is solved with [v2.27.0-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.27.0-beta). Thank you @ebaauw ebaauw

closed 2024-05-15 by pimw1