dresden-elektronik/deconz-rest-plugin #7687: Debounce device updates (Homekit Control)

unstressable, 2024-03-30:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Would be nice to have an option to debounce commands, because Homekit sends device updates very fast.

For example when changing a groups brightness from 1% to 100% Homekit sends somethin like 10%, 27%, 55%, 88% 100% very very quick.

It seems to cunfuse deconz, because some commands are lost after this operation.

Would be the best to have a checkbox to send commands only to deconz if it is more than about 500ms unchanged.

### Steps to reproduce the behavior

anything is described above

### Expected behavior

anything is described above

### Environment

Miniserver, Conbee 2, iobroker, deconz, Ubuntu, everythin up-to-date

Mimiix, 2024-03-30:
Hi feature requests are collected on the forums.

As this is not a bug as described in #5113 I am closing this issue

closed 2024-03-30 by Mimiix

ebaauw, 2024-03-30:
IMHO this should be handled by the API client, not by the API itself.  In Homebridge deCONZ, this is even configurable through `waitTimeUpdate`.