dresden-elektronik/deconz-rest-plugin #7689: MOES Smart Tür - und Fenstersensor

Feldi1606, 2024-04-01:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

MOES Smart Tür - und Fenstersensor

### Manufacturer

_TZ3000_8yhypbo7

### Model identifier

TS0203

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Feldi1606, 2024-04-01:
I hope I have included all the information that is necessary. If not, please report it.

Feldi1606, 2024-04-04:
Is there no one who can help me?

Mimiix, 2024-04-05:
@Smanar

Smanar, 2024-04-05:
It's a door sensor, you can try to add to this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0203_door_sensor.json

```
  "manufacturername": ["_TZ3000_n2egfsli", "_TZ3000_oxslv1c9", "_TZ3000_7tbsruql", "_TZ3000_7d8yme6f", "_TZ3000_rgchmad8", "_TZ3000_bzxlofth","_TZ3000_au1rjicn", "_TZ3000_4ugnzsli", "_TZ3000_decxrtwa", "_TZ3000_2mbfxlzr", "_TZ3000_6zvw8ham", "_TZ3000_v7chgqso", "_TZ3000_yxqnffam", "_TZ3000_zgrffiwg", "_TZ3000_8yhypbo7"],
  "modelid": ["TS0203", "TS0203", "TS0203", "TS0203", "TS0203", "TS0203","TS0203","TS0203","TS0203", "TS0203", "TS0203", "TS0203", "TS0203", "TS0203", "TS0203"],
```

Feldi1606, 2024-04-05:
I have tried the DDF an enter the ID in the header. The device will be shown in deconz and Phoscon but without an state.

[image]

and in FHEM the sensor is not shown.

Smanar, 2024-04-06:
>I have tried the DDF an enter the ID in the header

You have added the 2 values ? ModelID AND Manufacturename ?

If you can see it in phoscon, you need to see it in FHEM, bothapp are using the same API, perhaps need a synchronisation ? have you tried to restart the FHEM plugin ?
Perhaps the device will work in FHEM better than in phoscon (as it see a presense sensor ...) Else you can check direclty the api in phoscon/help/API Information/sensors

Feldi1606, 2024-04-06:
Yes i have added both Values. I delete the sensor and paired again.  Now i can see it in Deconz, Phoscon and Fhem but without any state.

> {
>     "config": {
>         "battery": 0,
>         "on": true,
>         "pending": [],
>         "reachable": true
>     },
>     "ep": 1,
>     "etag": "55030e4602669e988bdcee4467c99096",
>     "lastannounced": null,
>     "lastseen": "2024-04-06T19:04Z",
>     "manufacturername": "_TZ3000_8yhypbo7",
>     "modelid": "TS0203",
>     "name": "OpenClose 15",
>     "state": {
>         "lastupdated": "none",
>         "lowbattery": null,
>         "open": null,
>         "tampered": null
>     },
>     "type": "ZHAOpenClose",
>     "uniqueid": "a4:c1:38:71:5f:4a:52:b0-01-0500"
> }

Smanar, 2024-04-06:
I m searching information about your device on google, I don't found something "exotic". No tuya unlock sequence.

Perhaps the enrollement have failed ? Try to re-include it.
To make the enrollement step visible you an edit the "config/enrolled" part.

```
        {
          "name": "config/enrolled",
          "public": true
        },
```

Feldi1606, 2024-04-06:
Thank you very much for your help. 
I hope I was able to help you with the LOG. But I will send the devices back because I won't be able to continue testing in the next 3 weeks and I can't do what I want with the devices.

> 22:34:16:068 send permit join, duration: 65
> 22:34:18:068 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
> 22:34:28:068 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
> 22:34:28:512 skip create link for 0xE71A (lqi: 0) - 0xDA66 (lqi: 0)
> 22:34:28:778 APS-DATA.indication from unknown node 0x7797
> 22:34:28:780 CTRL create node 0xA4C138715F4A52B0, nwk: 0x7797
> 22:34:28:828 DEV Tick.Join: event/device.anounce
> 22:34:28:829 DEV Tick: fast poll 0xA4C138715F4A52B0, mac capabilities: 0x80
> 22:34:28:992 skip create link for 0xE71A (lqi: 0) - 0xFD72 (lqi: 0)
> 22:34:29:070 ZDP get node descriptor for 0x7797
> 22:34:29:278 DB pushZdpDescriptorDb()
> 22:34:29:280 DB UPDATE device_descriptors SET data = x'0240804111424200002a420000', timestamp = 1712435669 WHERE device_id = (SELECT id FROM devices WHERE mac = 'a4:c1:38:71:5f:4a:52:b0') AND endpoint = 0 AND type = 2
> 22:34:29:281 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'0240804111424200002a420000', 1712435669 FROM devices WHERE mac = 'a4:c1:38:71:5f:4a:52:b0'
> 22:34:29:318 ZDP get active endpoints for 0x7797
> 22:34:29:508 ZDP get simple descriptor 0x01 for 0x7797
> 22:34:29:642 DB pushZdpDescriptorDb()
> 22:34:29:644 DB UPDATE device_descriptors SET data = x'0104010204010401000300000500000803000400050006000800001019000a00', timestamp = 1712435669 WHERE device_id = (SELECT id FROM devices WHERE mac = 'a4:c1:38:71:5f:4a:52:b0') AND endpoint = 1 AND type = 4
> 22:34:29:646 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104010204010401000300000500000803000400050006000800001019000a00', 1712435669 FROM devices WHERE mac = 'a4:c1:38:71:5f:4a:52:b0'
> 22:34:29:979 sql exec SELECT sid FROM sensors
> 22:34:30:102 DEV found DDF for 0xA4C138715F4A52B0, path: /usr/share/deCONZ/devices/tuya/_TZ3000_TS0203_door_sensor.json
> 22:34:30:138 Websocket 127.0.0.1:54688 send message: {"attr":{"id":"9","lastannounced":null,"lastseen":"2024-04-06T20:34Z","manufacturername":"_TZ3000_8yhypbo7","mode":1,"modelid":"TS0203","name":"OpenClose 9","swversion":null,"type":"ZHAOpenClose","uniqueid":"a4:c1:38:71:5f:4a:52:b0-01-0500"},"e":"changed","id":"9","r":"sensors","t":"event","uniqueid":"a4:c1:38:71:5f:4a:52:b0-01-0500"} (ret = 2126909568)
> 22:34:30:148 Websocket 192.168.170.5:52099 send message: {"attr":{"id":"9","lastannounced":null,"lastseen":"2024-04-06T20:34Z","manufacturername":"_TZ3000_8yhypbo7","mode":1,"modelid":"TS0203","name":"OpenClose 9","swversion":null,"type":"ZHAOpenClose","uniqueid":"a4:c1:38:71:5f:4a:52:b0-01-0500"},"e":"changed","id":"9","r":"sensors","t":"event","uniqueid":"a4:c1:38:71:5f:4a:52:b0-01-0500"} (ret = 2126909568)
> 22:34:30:198 Websocket 127.0.0.1:54688 send message: {"e":"added","id":"9","r":"sensors","sensor":{"config":{"battery":0,"enrolled":null,"on":true,"pending":[],"reachable":null},"ep":1,"etag":"1b84f2198145361060b1903442a3b26e","id":"9","lastannounced":null,"lastseen":"2024-04-06T20:34Z","manufacturername":"_TZ3000_8yhypbo7","modelid":"TS0203","name":"OpenClose 9","state":{"lastupdated":"none","lowbattery":null,"open":null,"tampered":null},"type":"ZHAOpenClose","uniqueid":"a4:c1:38:71:5f:4a:52:b0-01-0500"},"t":"event","uniqueid":"a4:c1:38:71:5f:4a:52:b0-01-0500"} (ret = 2126909568)
> 22:34:30:213 Websocket 192.168.170.5:52099 send message: {"e":"added","id":"9","r":"sensors","sensor":{"config":{"battery":0,"enrolled":null,"on":true,"pending":[],"reachable":null},"ep":1,"etag":"1b84f2198145361060b1903442a3b26e","id":"9","lastannounced":null,"lastseen":"2024-04-06T20:34Z","manufacturername":"_TZ3000_8yhypbo7","modelid":"TS0203","name":"OpenClose 9","state":{"lastupdated":"none","lowbattery":null,"open":null,"tampered":null},"type":"ZHAOpenClose","uniqueid":"a4:c1:38:71:5f:4a:52:b0-01-0500"},"t":"event","uniqueid":"a4:c1:38:71:5f:4a:52:b0-01-0500"} (ret = 2126909568)
> 22:34:30:257 Websocket 127.0.0.1:54688 send message: {"config":{"battery":0,"enrolled":null,"on":true,"pending":[],"reachable":true},"e":"changed","id":"9","r":"sensors","t":"event","uniqueid":"a4:c1:38:71:5f:4a:52:b0-01-0500"} (ret = 2126909568)
> 22:34:30:262 Websocket 192.168.170.5:52099 send message: {"config":{"battery":0,"enrolled":null,"on":true,"pending":[],"reachable":true},"e":"changed","id":"9","r":"sensors","t":"event","uniqueid":"a4:c1:38:71:5f:4a:52:b0-01-0500"} (ret = 2126909568)
> 22:34:35:568 SC tick --> StateRead config/checkin, a4:c1:38:71:5f:4a:52:b0-01-0500
> 22:34:38:068 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
> 22:34:40:071 CTRL db store gui node a4:c1:38:71:5f:4a:52:b0
> 22:34:40:114 saved node state in 0 ms
> 22:34:40:116 sync() in 0 ms
> 22:34:40:593 SC tick --> StateRead config/checkin, a4:c1:38:71:5f:4a:52:b0-01-0500
> 22:34:46:069 SC tick --> StateRead config/checkin, a4:c1:38:71:5f:4a:52:b0-01-0500
> 22:34:46:158 [Master] read param with arg 0x19
> 22:34:46:238 CTRL read param resp id: 0x26, status: 0x04
> 22:34:48:068 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
> 22:34:51:569 SC tick --> StateRead config/checkin, a4:c1:38:71:5f:4a:52:b0-01-0500
> 22:34:54:471 APS-DATA.indication from unknown node 0x7714
[... cut ...]
> 22:34:58:725 	payload: 0b05290300
> 22:35:00:071 saved node state in 0 ms
> 22:35:00:074 sync() in 0 ms
> 22:35:02:568 SC tick --> StateRead config/checkin, a4:c1:38:71:5f:4a:52:b0-01-0500
> 22:35:08:068 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
> 22:35:09:838 max transmit errors for node 0xA4C138715F4A52B0, last seen by neighbors 41 s
> 22:35:11:488 max transmit errors for node 0xA4C138715F4A52B0, last seen by neighbors 42 s
> 22:35:11:603 SC tick --> StateRead config/checkin, a4:c1:38:71:5f:4a:52:b0-01-0500

Smanar, 2024-04-07:
There is nothing about enrollement 
`ry":0,"enrolled":0,"on`

But I have a doubt about the the "config/checkin" utility as your device don't have the poll cluster ...
For me you can remove 
```

        {
          "name": "config/checkin"
        },
```

closed 2024-05-06 by github-actions[bot]