dresden-elektronik/deconz-rest-plugin #7698: V2.26.3 don't start on macOS Sonoma 14.4.1

RP-Develop, 2024-04-07:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

V2.26.3 don't run on macOS Sonoma 14.4.1
[image]

### Steps to reproduce the behavior

Load the file from https://deconz.dresden-elektronik.de/macos/
Copied it to the program folder. 
Accept all questions from macOS.
Start the application.

### Expected behavior

After installation of v2.26.3 should run it.

### Environment

- Host system: MAC mini M2
- Running method: deConz on macOS
- Firmware version: 14.4.1
- deCONZ version (not Home assistant Addon version!): 2.26.3
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system?

RP-Develop, 2024-04-20:
Some new information: 

It was possible to let run it, but only if I unplugged the ConBee II and plugged it back in after startup.

On another system it was possible to run it with a ConBee III.

But now I also have other problems, independent of ConBee. The DDF files stored in /Users/home/Library/Application Support/dresden-elektronik/deCONZ/devices are not loaded. The DDF editor only shows "Untitled File".

I have to go back to 2.25.3.

aryelevin, 2024-05-15:
The new version 2.27.0 doesn't work either.
Any workaround?

manup, 2024-05-15:
Hmm that's odd, there is a bug fix for OpenSSL to fix a potential crash but otherwise it runs on my Mac Book Air.
Gonna check if I can reproduce it on M1 mini.

aryelevin, 2024-05-15:
The problem is about The DDF files stored in /Users/home/Library/Application Support/dresden-elektronik/deCONZ/devices are not loaded. The DDF editor only shows "Untitled File".

aryelevin, 2024-05-26:
Any updates on this?

RP-Develop, 2024-06-17:
...also 2.27.4 is crashed on macOS 14.5 during pairing.

manup, 2024-06-17:
Can you please download and try again, I have refreshed the package.
Digging deeper I found that the OpenSSL libraries where pointing to the wrong load location internally.

https://deconz.dresden-elektronik.de/macos/deconz_2.27.4-macos_x86_64.zip

RP-Develop, 2024-06-17:
sorry, but it crashed again:

v2.27.4
Qt: 5.15.2
GCC: 4.2.1 (C++ 201402)
OpenSSL: 3.3.0
Commit: d9ddb8
Firmware: 0x26510900

-------------------------------------
Translated Report 
-------------------------------------

Process:               deCONZ [58172]
Path:                  /Applications/deCONZ.app/Contents/MacOS/deCONZ
Identifier:            de.phoscon.deconz
Version:               2.27.4 (???)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2024-06-17 19:53:17.0203 +0200
OS Version:            macOS 14.5 (23F79)
Report Version:        12
Anonymous UUID:        78AE6119-A7D7-BC0D-8DC1-8D94E068485D

Sleep/Wake UUID:       B8C521A6-E0F7-44F7-950D-708276A558BA

Time Awake Since Boot: 720000 seconds
Time Since Wake:       16068 seconds

System Integrity Protection: enabled

Notes:
PC register does not match crashing frame (0x0 vs 0x1735AC272)

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000060
Exception Codes:       0x0000000000000001, 0x0000000000000060

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [58172]

VM Region Info: 0x60 is not in any region.  Bytes before following region: 4305338272
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1009e4000-100ad0000    [  944K] r-x/r-x SM=COW  /Applications/deCONZ.app/Contents/MacOS/deCONZ

Error Formulating Crash Report:
PC register does not match crashing frame (0x0 vs 0x1735AC272)

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libde_rest_plugin.dylib       	       0x1735ac272 DeviceDescriptions::get(Resource const*, DDF_MatchControl) + 306
1   libde_rest_plugin.dylib       	       0x1735ace9a DeviceDescriptions::handleDDFInitRequest(Event const&) + 154
2   QtCore                        	       0x10cc00f95 0x10c9d5000 + 2277269
3   libde_rest_plugin.dylib       	       0x173534af4 EventEmitter::eventNotify(Event const&) + 52
4   libde_rest_plugin.dylib       	       0x173699eda EventEmitter::process() + 618
5   libde_rest_plugin.dylib       	       0x173645ec2 DeRestPluginPrivate::apsdeDataIndication(deCONZ::ApsDataIndication const&) + 2210
6   QtCore                        	       0x10cc010d5 0x10c9d5000 + 2277589
7   libdeCONZ.1.2.0.dylib         	       0x109743de7 deCONZ::ApsController::apsdeDataIndication(deCONZ::ApsDataIndication const&) + 55
8   deCONZ                        	       0x100a47d1d zmController::onApsdeDataIndication(deCONZ::ApsDataIndication const&) + 12285
9   deCONZ                        	       0x100a698ad zmMaster::processPacked(zm_command const*) + 4781
10  deCONZ                        	       0x100a6d9d2 0x1009e4000 + 563666
11  deCONZ                        	       0x1009f440a protocol_receive + 282
12  deCONZ                        	       0x100a6d4d8 SerialCom::readyRead() + 648
13  deCONZ                        	       0x100a6d0fe 0x1009e4000 + 561406
14  QtCore                        	       0x10cbf8b9f QObject::event(QEvent*) + 943
15  QtWidgets                     	       0x10a0ef9ea QApplicationPrivate::notify_helper(QObject*, QEvent*) + 266
16  QtWidgets                     	       0x10a0f0e11 QApplication::notify(QObject*, QEvent*) + 497
17  QtCore                        	       0x10cbcda34 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
18  QtCore                        	       0x10cbced79 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 809
19  libqcocoa.dylib               	       0x10aeea259 0x10aeb0000 + 238169
20  libqcocoa.dylib               	       0x10aeea9c8 0x10aeb0000 + 240072
21  CoreFoundation                	    0x7ff8016617d6 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
22  CoreFoundation                	    0x7ff801661779 __CFRunLoopDoSource0 + 157
23  CoreFoundation                	    0x7ff801661548 __CFRunLoopDoSources0 + 215
24  CoreFoundation                	    0x7ff8016601b8 __CFRunLoopRun + 919
25  CoreFoundation                	    0x7ff80165f859 CFRunLoopRunSpecific + 557
26  HIToolbox                     	    0x7ff80c552a09 RunCurrentEventLoopInMode + 292
27  HIToolbox                     	    0x7ff80c552816 ReceiveNextEventCommon + 665
28  HIToolbox                     	    0x7ff80c552561 _BlockUntilNextEventMatchingListInModeWithFilter + 66
29  AppKit                        	    0x7ff804c31c61 _DPSNextEvent + 880
30  AppKit                        	    0x7ff805545dc0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1273
31  AppKit                        	    0x7ff804c23075 -[NSApplication run] + 603
32  libqcocoa.dylib               	       0x10aee962f 0x10aeb0000 + 235055
33  QtCore                        	       0x10cbc9acf QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
34  QtCore                        	       0x10cbce042 QCoreApplication::exec() + 130
35  deCONZ                        	       0x1009fc9ae main + 5838
36  dyld                          	       0x200c98366 start + 1942

manup, 2024-06-17:
Interesting this log is very helpful. For one it shows that OpenSSL is working but and also the function where it crashes.
I currently don't see why it crashes in there though.

Which device did you try to join? Perhaps I can re-create the crash.

RP-Develop, 2024-06-17:
I have tried the IKEA Badring E2202 and the IKEA Shortcut Button E1812. Only these devices were tried. It is my test system on a MacBook Air M1 + ConBee III + v2.27.4, it is not my productive system. My production system works with version 2.25.3 on a Mac Mini M2 and with ConBee II. Unfortunately, V2.25.3 is the last version I had running.

manup, 2024-06-19:
Wow this was really nasty to debug, after a while I could recreate the crash in my macOS setup (2018 Intel MacBook).
With a small fix the Ikea switch does now join correctly. Note this wasn't bound to Ikea specifically but to any device for which no DDF is currently loaded before pairing.

Can you please try following version:

https://deconz.dresden-elektronik.de/macos/deconz_2.27.4-macos_x86_64_6f307e.zip

RP-Develop, 2024-06-19:
...it doesn't look bad, I'll need some time to test everything. I've currently programmed a TRETAKT socket and a BADRING without a crash. Unfortunately I'm not getting a water message from the BADRING, even though it shows ENROLLED in the IAS Zone cluster. This has already been covered in another issue, but unfortunately without success for me.

manup, 2024-06-24:
I guess the BADRING is a separate issue not related to the macOS version.
How is the setup holding up so far, can we close this issue?

manup, 2024-07-08:
Closing for now assuming the fix works.

closed 2024-07-08 by manup