dresden-elektronik/deconz-rest-plugin #7703: Moes Smart Dimmer Module 1 Gang ZM-105-M

dmichalski, 2024-04-09:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Moes Smart Dimmer Module 1 Gang ZM-105-M

### Manufacturer

_TZE204_dcnsggvz

### Model identifier

TS0601

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

dmichalski, 2024-04-09:
The dimmer is recognized by Phoscon but wrongly identified as a plug instead of a dimmable light. Anything else I can do to get that one added?

Smanar, 2024-04-10:
Hello can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_dcnsggvz",
  "modelid": "TS0601",
  "vendor": "$MF_TUYA",
  "product": "I gang dimmer",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_DIMMER_SWITCH",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "none"},
        },
        {
          "name": "state/bri",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = (Attr.val / 1000.0) * 254.0;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "(Item.val / 254.0) * 1000.0;"},
          "read": {"fn": "none"},
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}
```

I can be wrong for tuya dpid, have made this DDF with another one.

dmichalski, 2024-04-13:
Thanks for the DDF! 

Unfortunately I'm not able to get this one working properly. I placed the file inside ~/.local/share/dresden-elektronik/deCONZ/devices and did a systemctl deconz restart. But the dimmer is still recognized as Smart Plug and not as light. Am I missing out on something? Sorry, first time I'm working with DDF.

Btw. is there a way to really edit the DDF in deconz GUI? I see the menu item and can open the edit pane. But it seems to be read only...

Smanar, 2024-04-13:
> But the dimmer is still recognized as Smart Plug and not as light

The difference will not be visible in deconz, but on the API.
Deconz show the reality, it's from the device firmware, but in the API, the device will be reconised as 

`"type": "$TYPE_DIMMER_SWITCH",`

IDK what is your third app ? else can try with phoscon/help/API Information/light

>Btw. is there a way to really edit the DDF in deconz GUI? I see the menu item and can open the edit pane. But it seems to be read only

It's possible, but not easier, almost for tuya.

dmichalski, 2024-04-13:
Ok, nevermind. The JSON above has two trailing "," on the item descriptions which led to an error during the file import. So for future visitors, the following version works like a charm for the Moes dimmer:

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE204_dcnsggvz",
  "modelid": "TS0601",
  "product": "TS0601",
  "sleeper": false,
  "status": "Gold",
  "path": "/devices/ts0601.json",
  "subdevices": [
    {
      "type": "$TYPE_DIMMER_SWITCH",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"},
          "refresh.interval": 86400
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = Attr.val;" },
          "write": {"fn": "tuya", "dpid": 1, "dt": "0x10", "eval": "Item.val == 1 ? 1 : 0;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/bri",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = (Attr.val / 1000.0) * 254.0;"},
          "write": {"fn": "tuya", "dpid": 2, "dt": "0x2b", "eval": "(Item.val / 254.0) * 1000.0;"},
          "read": {"fn": "none"}
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ]
}

Thanks again @Smanar!

Smanar, 2024-04-13:
Do you want to make a PR to add officialy or want I make it ?

Mimiix, 2024-04-13:
Please dont close unless a PR has been merged.

dmichalski, 2024-04-13:
Just created the PR. Hope I did everything right...

Smanar, 2024-04-14:
For me it's fine, you have a strange error, but I don't see the problem, I m asking.

closed 2024-04-22 by manup