dresden-elektronik/deconz-rest-plugin #7709: Rain Seer (Tuya) water valve -TS0049 by _TZ3210_0jxeoadc

Addamka84, 2024-04-14:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Rain Seer

### Manufacturer

_TZ3210_0jxeoadc

### Model identifier

TS0049

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

In ZHA its been implemented (if its help): https://github.com/zigpy/zha-device-handlers/issues/2377

dlondero, 2024-04-30:
Would be great to have this added!

kkossev, 2024-05-01:
@ devs - pay attention to the fact that this TS049 device requires Tuya cluster command 0x04 (instead of the usual command 0x00) for the open/close commands, setting the irrigation timer, etc.

closed 2024-05-30 by github-actions[bot]

yulasinio, 2024-07-12:
Is there any chance that this device can be added to ZHA? I just recently purchased one and would be great to be supported nativelly so I don't have to install Z2M