dresden-elektronik/deconz-rest-plugin #7710: Tuya 4CH Zigbee Switch Module-L TS0014 _TZ3000_cxw02yp3

janstenpickle, 2024-04-18:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

4CH Zigbee Switch Module-L

### Manufacturer

_TZ3000_cxw02yp3

### Model identifier

TZ-0014

### Device type to add

Light

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

closed 2024-05-23 by github-actions[bot]

janstenpickle, 2024-05-25:
@Smanar sorry for the delay replying to you, I've only just had a chance to try this again. Thank you so much for this, it worked perfectly! 👌

Smanar, 2024-05-25:
Nice, do you to make the PR ?
Or want I make it ? to have it officialy.