dresden-elektronik/deconz-rest-plugin #7711: Issues in setting Philps Hue Outdoor Sensor SML004 sensitivity

mariomaz87, 2024-04-18:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I'm trying to change the sensitivity of my Philips Hue Outdoor Motion Sensor, version SML004.
Using the Deconz UI I can see that Sensitivity is 0. The ID of the attribute is 0x0030 and it's the same ID of "Physical Contact Occupied to Unoccupied" attribute. If I try to change the sensitivity using Hue Essential or HA developers tool with command:

service: deconz.configure
data:
  entity: binary_sensor.presence_17
  field: /config
  data:
    sensitivity: 2

I see that Physical Contact Occupied to Unoccupied attribute changes to 2, while Sensitivity is always 0.

### Steps to reproduce the behavior

In HA Developer Tools, Services:

service: deconz.configure
data:
  entity: binary_sensor.presence_17
  field: /config
  data:
    sensitivity: 2

### Expected behavior

Change Sensitivity attribute to specified value

### Screenshots

[image]

### Environment

- Host system: Raspberry Pi
- Running method: Marthoc Docker container
- Firmware version: 26780700
- deCONZ version (not Home assistant Addon version!): 2.26.3
- Device: ConBee II
- Do you use an USB extension cable: tes
- Is there any other USB or serial devices connected to the host system? If so: Which? No

ebaauw, 2024-04-18:
As you can see from the screenshot, they messed up `general.xml` and defined another set of manufacturer-specific attributes with the same IDs (0x0030 and 0x0031) as the Hue attributes.  You need to delete these, or move the Hue-specific attributes before them, for the GUI to handle the Hue-specific attributes correctly.

mariomaz87, 2024-04-19:
> As you can see from the screenshot, they messed up `general.xml` and defined another set of manufacturer-specific attributes with the same IDs (0x0030 and 0x0031) as the Hue attributes. You need to [...]

Thanks. Is this something I can do in the GUI? Do I have to change some configuration file? Thanks!

ebaauw, 2024-04-19:
You need to edit `general.xml` and restart deCONZ.  Typically it's in `/usr/share/deCONZ/zcl/`.  Best to make a backup, but it's also restored when re-installing deCONZ.

mariomaz87, 2024-04-19:
I'm using docker so I think it's a little bit different. I copied the file from the docker container, using docker cp, to a local folder, I removed the whole section for "Physical Contact Configuration" and copied the file back in the container, I rebooted but if I log with VNC I still see the same old configuration. I double checked and the general.xml is the updated one.

ebaauw, 2024-04-19:
Sensitivity is from 0 (least sensitive) to the value of _Sensitivity max_, 4 (most sensitive).

mariomaz87, 2024-04-20:
Thanks. I'm already using usertest and I confirm it skips any delays. Any sensitivity setting below 2 prevents the sensor from detecting any movement. 2, 3, and 4 work as expected increasing the sensitivity, but with 1 and 0 the sensor stops working. 
I remember that while using the Hue hub it was possibile to set the sensitivity from 1 to 5, so it's a little bit strange that in Deconz I only have 3 useful settings.

ebaauw, 2024-04-20:
This has nothing to do with deCONZ; sensitivity is handled by the sensor itself.

Mimiix, 2024-04-20:
In that case I'm closing this issue.

closed 2024-04-20 by Mimiix