dresden-elektronik/deconz-rest-plugin #7713: Tuya Zigbee Photoelectric Smoke Fire Alarm Detector

swissmaster1, 2024-04-20:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya Zigbee Photoelectric Smoke Fire Alarm Detector 

### Manufacturer

_TZE200_rccxox8p

### Model identifier

 TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

swissmaster1, 2024-04-20:
The Sensor looks identical to this one here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6734
I tried the DDF listed there from the user [BabaIsYou](https://github.com/BabaIsYou), but it didnt work.

Smanar, 2024-04-23:
It's the one you have tested ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6734#issuecomment-1423453064
Try setting the status to "Gold".

The device use the dpid = 1 for fire, so his DDf need to work, you have replaced manufacturername AND modelid ?

This device is locked, need to add too (and re-include it after, the unlock sequence is done at inclusion or every 24)
```
        {
          "name": "config/tuya_unlock"
        }
```
For battery
```

        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 15, "eval": "Item.val = Attr.val;"},
          "read": {"fn": "none"}
        },
```

swissmaster1, 2024-04-23:
Thanks for your help.
Correct like this?

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_rccxox8p",
  "modelid": "TS0601",
  "vendor": "Tuya",
  "product": "Photoelectric Smoke sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0xef00"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 15, "eval": "Item.val = Attr.val;"},
          "read": {"fn": "none"}
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/errorcode",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 11,
            "eval": "Item.val = String(Attr.val);",
            "fn": "tuya"
          },
          "default": "0"
        },
        {
          "name": "state/fire",
          "awake": true,
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = (Attr.val == 1);",
            "fn": "tuya"
          },
          "default": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/test",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 101,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": 0
        },
		{
          "name": "config/tuya_unlock"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    }
  ]
}`

swissmaster1, 2024-04-23:
Perfect it worked. The sensor is now detected by Phoscon. This is what home assistant shows now:

[image]

Is that correct? Isn't the status wrong?

Smanar, 2024-04-24:
>I do not understand what you mean by reinclude it? It's not enough how I added it in the DDF i posted before? Do I have to add it somewhere else? Sorry I'm very now to Deconz and DDF stuff :-)

Reset the device and re include in in the zigbee network.
Your DDF is perfect, it's just the deconz working mode, you see only
```
	{
          "name": "config/tuya_unlock"
        }
```

But deconz see 

```
{
  "schema": "resourceitem1.schema.json",
  "id": "config/tuya_unlock",
  "datatype": "Bool",
  "access": "R",
  "public": false,
  "description": "Tuya sequence to unlock some device.",
  "default": false,
  "parse": {
    "fn": "zcl:attr",
    "ep": 1,
    "cl": "0x0000",
    "at": "0xfffe",
    "eval": "Item.val = (Attr.val != 0)"
  },
  "read": {
    "fn": "zcl:attr",
    "ep": 1,
    "cl": "0x0000",
    "at": [
      "0x0004",
      "0x0000",
      "0x0001",
      "0x0005",
      "0x0007",
      "0xfffe"
    ]
  },
  "refresh.interval": 86400
}
```

It's a request sequence to unlock the device, and deconz NEED to make it to unlock the device (only 1 time, after it's ok).
But there is nothing to trigger deconz, so it make it the first time he see the device and every 24h `"refresh.interval": 86400`.
It's just a "cheat" to force deconz to make the sequence.

To resume, don't worry, your DDF is nice, just if an user have issue with it, just say him to re-include the device.

closed 2024-05-16 by swissmaster1

dmichalski, 2024-05-26:
I happen to have obtained this detector aswell and tried to include via the DDF in this thread. Unfortunately I can't get the detector to report the fire alarm properly. It's constantly reporting fire. The test mode is detected fine if I press the Test button on the device.
 
My guess is that the dpid for fire is maybe different on my device?! To confirm that - can anybody tell me how to check out the dpids of the device? I can't figure it out by looking at the cluster infos. "Tuya specific" is empty and on the "Basic" cluster I see the same as on the screenshots above (which would lead me into thinking that the dpid is equivalent to the id and therefore it should be id 13 as it's called "alarm mask).

Smanar, 2024-05-26:
For tuya, you can just enable the logging, Deconz/help/debug view with flag "info" and "info_l2" you will see all Tuya requests from and to the device and their dpid.

phuongpham, 2024-07-20:
> I happen to have obtained this detector aswell and tried to include via the DDF in this thread. Unfortunately I can't get the detector to report the fire alarm properly. It's constantly reporting [...]

I have the same issue. The binary switch is reporting ON all the time.
@swissmaster1 could you please share you latest DDF file?
I also have the problem with the test switch not displaying in my openHAB overview

[image]

Smanar, 2024-07-20:
>I have the same issue. The binary switch is reporting ON all the time.

If you can take a look on logs for a missed request ?

>I also have the problem with the test switch not displaying in my openHAB overview

Can you see it direclty in the API ? Using phoscon/help/API Information/sensors.

phuongpham, 2024-07-20:
> > I have the same issue. The binary switch is reporting ON all the time.
> 
> If you can take a look on logs for a missed request ?
> 
> > I also have the problem with the test switch not displaying in my openHAB overview
> 
> Can you see it direclty in the API ? Using phoscon/help/API Information/sensors.

Yes I get this from the API

{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true
    },
    "etag": "XXXXX",
    "lastannounced": null,
    "lastseen": "2024-07-20T13:19Z",
    "manufacturername": "_TZE200_rccxox8p",
    "modelid": "TS0601",
    "name": "Thermostat",
    "state": {
        "errorcode": "false",
        "fire": false,
        "lastupdated": "2024-07-20T13:19:28.087",
        "test": false
    },
    "swversion": "1.0.8",
    "type": "ZHAFire",
    "uniqueid": "XXXXX"
}

Smanar, 2024-07-21:
So for the second problem, it's probably because the OpenHAB plugin don't support this field. Not a big problem, you can ask the dev to add it, but IDK if this field is so usefull ?

The first one is more problematic, I don't know this device, but some alarm device send only a notification on alarm, so they are always "on", but make a new notfication at every detection.

swissmaster1, 2024-07-21:
> I happen to have obtained this detector aswell and tried to include via the DDF in this thread. Unfortunately I can't get the detector to report the fire alarm properly. It's constantly reporting [...]

[tuya_smoke_sensor.json](https://github.com/user-attachments/files/16323951/tuya_smoke_sensor.json)

This is my DFF for the smoke sensore. Until now it seems to work fine.