dresden-elektronik/deconz-rest-plugin #7720: Aqara Dual Relay Module T2

moddavid, 2024-04-21:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Aqara Dual Relay Module T2 (DCM-K01)

### Manufacturer

Xiaomi Aqara LUMI

### Model identifier

lumi.switch.acn047

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

### Electric Measurement
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Simple Metering
[image]
[image]
[image]

### Multistate Input (Basic)
[image]

### On/Off
[image]
[image]

### Lumi Specific
[image]
[image]
[image]
[image]
[image]

### Analog Input (Basic)
[image]

### Additional information
Phoscon shows it currently as two On/Off lights. I have installed the T2 behind my existing roller shader switch to make it smart. While moving the shader up by activating the corresponding "light" worked, the other "light" for the down movement didn't do anything.
I'm running the docker image deconzcommunity/deconz:2.26.3 on an Intel NUC with Conbee II.

SwoopX, 2024-04-23:
Hi there.

Unfortunately, the screenshots indicate that the clusters have not been read properly. [This page](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#read-device-clusters) explains how to read clusters. Please take new ones and update the issue accordingly. Thanks a lot!

gee-jay-bee, 2024-04-24:
Screenshots taken in order shown (hope this is correct …):

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

moddavid, 2024-05-05:
@SwoopX sorry for the delay, here are the new screenshots:

Node Info
[image]

Endpoints and clusters
[image]

Basic
[image]
[image]

Electric Measurement
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Simple Metering
[image]
[image]
[image]

Multistate Input (Basic)
[image]

On/Off
[image]
[image]

Lumi Specific
[image]
[image]
[image]
[image]
[image]

Lumi Specific (second device - this is the only one which was different)
[image]
[image]
[image]
[image]
[image]

Analog Input (Basic)
[image]

This time both up and down movement worked just fine, I'm not sure why it didn't work the last time.

moddavid, 2024-05-27:
Anything we can do here?
For me right now two options are missing: Measuring the power consumption (like the Aqara smart plug SP-EUC01 - 
lumi.plug.maeu01) and the child lock (like the Aqara Thermostat E1 - lumi.airrtc.agl001).

gee-jay-bee, 2024-06-18:
Fwiw, I was experiencing some challenges with Deconz + Conbee II and finally took a chance on Zigbee2mqtt with a newer USB coordinator. This T2 is natively supported there and good news is the drop-out issue I was seeing isn't happening. There is power reporting (though it seems to be the combined power of both relays) but I don't see child lock. You can disconnect the relays from the inputs.

Deconz served me well for many years and I hope the new DDF strategy can accelerate device on-boarding. Good luck!

moddavid, 2024-06-18:
I would rather don't stale this ticket for now. Maybe I find some time to compare the DDF files from the mentioned other devices to see if I get it somehow to work as intended.

SwoopX, 2024-07-20:
I'll share a DDF candidate within the next days

closed 2024-08-08 by manup

sch115, 2024-09-05:
Hello to all!
I ran into a issue with energy consumption statistics. Voltage and watts numbers are OK but Total Consumption in kWh is 1000 time higher. I believe this link was based on the same problem as mentioned: https://github.com/Koenkk/zigbee2mqtt/issues/22148

Current versions:
Raspbian GNU/Linux Bullseye (11)
Homebridge 1.8.4
deCONZ version 2.28.1
ConbeeII 26780700
lumi.switch.acn047 0.0.0_0030

Everything else looks good to me.
Thank you for your help!