dresden-elektronik/deconz-rest-plugin #7728: Unable to discover Namron Zigbee Touch thermostat 16A 2.0 (modelid 4212752)

isaksengeir, 2024-04-26:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I am running home assistant with a raspbee2 and deconz add-on. It is unable to discover the New Namron Zigbee Touch thermostat 16A 2.0 (modelid 4212752). I have updated to latest firmware. I believe the issue is related to ddf. Going into /usr/share/deCONZ/devices/namron I see 4512737_thermostat.json that supports modelid 4512737 and 4512738 (https://www.elektroimportoren.no/docs/lib/4512737-FDV-1.pdf). The namron thermostats I recently purchased are quite new and have modelid 4512752 (https://www.elektroimportoren.no/docs/lib/4512752-FDV-1.pdf). I attempted to create a copy of 4512737_thermostat.json and renmed it to 4512752_thermostat.json and changed the and updated the modelid field to 4512752. The rest of the ddf I kept the same. I restarted the gateway and home assistant. But still not able to find the thermostat. 

### Steps to reproduce the behavior

Purchase the New Namron Zigbee Touch thermostat 16A 2.0 (modelid 4212752). Install home assistant OS on a raspberry pi with a raspbee 2 zigbee gateway and install the deCONZ add-on. Put thermostat in discover mode and follow the instructions from the "add thermostat" in Phoscon.

### Expected behavior

Nothing will happen. It will not discover the thermostat. 

### Environment

- Host system: Raspberry Pi 
- Running method: Home Assistent deCONZ Add-on 
- Firmware version: (26780700)
- deCONZ version (not Home assistant Addon version!): (2.25.3)
- Device: RaspBee II

isaksengeir, 2024-04-27:
Sorry, I typed a wrong digit in the model id here when reporting. The correct is 4512752

isaksengeir, 2024-04-27:
I just realised that making a new ddf or editing existing ddf inside the container will not survive restarting deCONZ and the continuer. Thus, the request is probably an update to the container rather than a bug. But, I could be wrong.

Mimiix, 2024-04-27:
Hi,

Is it showing in the deconz Gui as hex number?

Can you share a screenshot of your full deconz Gui?

If a device is not supported, it shows as hex number.

isaksengeir, 2024-04-27:
[image]

Mimiix, 2024-04-27:
> ![Screenshot 2024-04-27 at 12 14 28](https://private-user-images.githubusercontent.com/6112900/326171094-fb9938f2-20a8-43a6-ad94-885f4eaf59c5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyMjUyNDksIm5iZiI6MTcxNDIyNDk0OSwicGF0aCI6Ii82MTEyOTAwLzMyNjE3MTA5NC1mYjk5MzhmMi0yMGE4LTQzYTYtYWQ5NC04ODVmNGVhZjU5YzUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjdUMTMzNTQ5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NWVlNTgzNmUzYjNiNjhlZDUwYWVmZTFlNWE3ZGQyYjkwZTljYWM4NmRkNzQxOTM1NjkyNTVjMWVjZDI2YWFkZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.YDlih4oW_NOQk8HgQ9d9hxM8NIJ5b5_8LLsgDD9l95g)

I see 2 hex named devices on the left top. Can you read the basic clusters and share the screenshots?

isaksengeir, 2024-04-27:
[image]

isaksengeir, 2024-04-27:
[image]

Mimiix, 2024-04-27:
Hi this is the node info, not basic cluster. In the Wiki of this GitHub is a go-to on how to do it.

isaksengeir, 2024-04-27:
Sorry about that. Here are the basic cluster for the two:
[image]
[image]

Mimiix, 2024-04-27:
Can you please read them? These are not read. You need to click the read button. It helps if you click the button on the device to wake it up.

isaksengeir, 2024-04-27:
@Mimiix they only blink red, so I think it is a dead end. I also believe that these two were there before I purchased the namron zigbee thermostats. Thus, they are simply not showing up here (I think).

Mimiix, 2024-04-28:
> @Mimiix they only blink red, so I think it is a dead end. I also believe that these two were there before I purchased the namron zigbee thermostats. Thus, they are simply not showing up here (I think).

Are those the same type?

isaksengeir, 2024-04-28:
I am not sure what you refer to with same type here. I have purchase a total of 6 namron zigbee thermostats and none of them shows up in phoscon/deconz. My guess would be that the are simply not supported yet and that that this should be an easy fix since the two previous zigbee tech thermostat version integrates well.

Mimiix, 2024-04-28:
> I am not sure what you refer to with same type here. I have purchase a total of 6 namron zigbee thermostats and none of them shows up in phoscon/deconz. My guess would be that they are simply not supported yet and that that this should be an easy fix since the two previous zigbee tech thermostat version integrates well.

If they are all 6 the same, and you are saying those 2 are the same type you tried to pair: there is no bug here and indeed they are not supported and there is not issue so far.

You need to file a device request with screenshots.

If you can't pair them, you need to ask on the forums for that.

Closing as this is not a bug as described in #5113

closed 2024-04-28 by Mimiix

isaksengeir, 2024-04-28:
First of all, I am grateful for your replies, but I do not understand why you close this issue. I do not think you have understood the problem either. The 2 grey boxes with hex you refer to has nothing to do with this, they were there from before. 

The issue is that the newest namron zigbee touch thermostats are not discovered by deconz/phoscon. I thought this was a simple DDF fix, but I have not managed to mace deconz discover the by editing ddf either. Anyway, thank you for taking the time. I hope someone will release a fix for this.

Mimiix, 2024-04-29:
> First of all, I am grateful for your replies, but I do not understand why you close this issue. I do not think you have understood the problem either. The 2 grey boxes with hex you refer to has [...]

Hi,

If they are not discoverable, it either has something to do with the firmware of the device or with your network. Devices should _always_ show up in deCONZ regardless deCONZ containing support. A DDF can only be used when it is connected to the network.

To find out if its your network or the firmware, i recommend posting on the forums as that is not covered on this github.