dresden-elektronik/deconz-rest-plugin #7734: to invert the shutter state

sanluca, 2024-04-30:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hi guys,
I'm using a _TZ3000_iaxvag8w roller shutter module produced by TS130F. However, I'm facing an issue with OpenHAB. The roller shutter module indicates a fully open state as 100% and a closed state as 0%. Conversely, OpenHAB interprets the states differently: 0% indicates an open shutter and 100% indicates a closed one. Is there a way to invert the state of the Zigbee module?
thanks

### Steps to reproduce the behavior

.

### Expected behavior

.

### Environment

- Host system: (ubuntu)
- Running method: (Ubuntu/openhab)
- Firmware version: (26720700)
- deCONZ version (not Home assistant Addon version!): (2.26.3)
- Device: (ConBee II)

Mimiix, 2024-04-30:
@Smanar is this a bug or feature request?

sanluca, 2024-04-30:
> @Smanar is this a bug or feature request?

I'm not sure if it's supposed to work this way or if there needs to be an option to invert the values. I don't have any other products to compare it to

Smanar, 2024-04-30:
I don't see this device on DDF or legacy code.
So I think we can create a DDF to reverse the value returned by deconz if the problem is from deconz.

On deconz rules are 
```
shutter closed
on = true
bri = 255
open = false
lift = 100%
```

I don't now if the HA plugin do a reverse.
It's a module, so you have wired command, thoses one work in correct side ?
Can you check in the API if deconz return the good one ? For exemple in phoscon/help/API Information/light.

If value are reversed, can you share the auto generated DDF on the editor (on text version, not picture), else the complete  json from the API you have get just before ?

With DDF I will be able to return the value returned, but not the requests made.

Mimiix, 2024-04-30:
In that case I'm going to ask for a device request @sanluca

closed 2024-04-30 by Mimiix