dresden-elektronik/deconz-rest-plugin #7741: Problems with Vimar Window Covering device

LeoeLeoeL, 2024-05-04:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I have 3 Vimar window covering devices paired to Conbee II.
Until today they worked OK.
Today one of them didn't respond to OPEN commands even if sometimes the circle was blinking blue.
I resetted by F5 and now I have:
[image]

Normally it's:
[image]

### Steps to reproduce the behavior

Press F5 and wait

### Expected behavior

I expect device working as usual

### Environment

- Host system: Raspberry Pi 3b
- Running method: Raspbian
- Firmware version: 0x26780700
- deCONZ version (not Home assistant Addon version!): v2.26.2-beta
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? NO

Mimiix, 2024-05-04:
Hi,

As this is not a bug report as described in #5113 I am closing this issue.

I recommend opening an topic on the forums.

Kind regards,

closed 2024-05-04 by Mimiix

LeoeLeoeL, 2024-05-04:
And what is it?

Smanar, 2024-05-04:
For me it's an inclusion problem.
Your device is not full included (the 0xffff), try to re-include it, without delting it.

LeoeLeoeL, 2024-05-04:
There is no way to fix that manually.
Even resetting the device from the device (very very annoying method). I also paired the device to Zigbee2Mqtt (using another coordinator) and then paired again to Deconz.
Same picture as above.
I had something similar few week ago.
Deconz fixed that after 4 or 5 days. The difference was that last time the device didn't work until deconz fixed it.

Mimiix, 2024-05-04:
Guys, please proceed on the forums.