dresden-elektronik/deconz-rest-plugin #7754: Tuya Smart Zigbee Smoke Detector, Model PST-YG500A: Device request

Winnie-LD, 2024-05-09:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya Smart Zigbee Smoke Detector, Model PST-YG500A

### Manufacturer

Tuya

### Model identifier

ConBee II

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Winnie-LD, 2024-05-11:
Ist this sufficient information for covering the device? Do you need further information / screenshots?
thank you for your support!

Mimiix, 2024-05-11:
> Ist this sufficient information for covering the device? Do you need further information / screenshots? thank you for your support!

This is not the smoke detector, this is the Conbee.

Winnie-LD, 2024-05-12:
Ah, ok. I see. How do I extract information from the smoke detector? Every time I run deconz on my PC (with ConBee II attached), it would crash as soon as I enable "join network".

Winnie-LD, 2024-05-12:
A simple reinstall of the deconz software on Win11 did the job...
Now I can see some of my devices, unfortunately not the smoke sensor (why should I, if it doesn't connect...?)
What do I do now, in order to fetch necessary informations from a device that won't show up or connect?

Mimiix, 2024-05-12:
> A simple reinstall of the deconz software on Win11 did the job... Now I can see some of my devices, unfortunately not the smoke sensor (why should I, if it doesn't connect...?) What do I do now, in order to fetch necessary informations from a device that won't show up or connect?

It should show up as a hex name in deconz. You won't see it in phoscon.

Winnie-LD, 2024-05-12:
Yes, it does. And I was able to pair it in Phoscon App. However, it does not deliver useful information.
How do I get it to reveal its readings?

Smanar, 2024-05-12:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

I need at least attribure from basic cluster 0x0000 (rememeber you need to awake the device)
And the full node picture with cluster visible

Mimiix, 2024-05-12:
Don't post them as a comment, please update the original post here

Winnie-LD, 2024-05-12:
OK, thank you once more. I updated the first post, is this helpful? Do you need more / other information?

Smanar, 2024-05-13:
Yes it's fine, and I have a good new, will be easy for you, your device is already supported ^^

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3210_up3pngle_TS0205_smoke_sensor.json

It's for that the name is "fire 4" and not "0xXXXX", it's probably your fourth try to include it. Can try it, but for me all is fine.

Winnie-LD, 2024-05-13:
I tried this ddf before and deconz would recognize the smoke sensor but as a YG400A of course. The device does not deliver any useful information, trigger, events or whatever you call it. It shows a nice symbol (a flame) but that's all.
I guess, I need a specific ddf for this device?

Smanar, 2024-05-13:
>  but as a YG400A of course

Useless, it's tuya, it's a clone of clone of clone, it's just "one more clone" of a smoke sensor.

>The device does not deliver any useful information, trigger, events or whatever you call it. It shows a nice symbol (a flame) but that's all

You mean in Phoscon ? Try to take a look in phoscon/help/API Information and tab "event". Now try to trigger the device.

Winnie-LD, 2024-05-13:
Ah, that's it. I see it recognizes smoke (already tested it) and the alarm sets off.
The "fire" reading jumps from 0 to 1 - that should be the trigger I was looking for.

But as far as I can see, I can't control the siren via zigbee?

Smanar, 2024-05-14:
>But as far as I can see, I can't control the siren via zigbee?

IDK, from Z2M the _TZ3210_up3pngle is a YG400A, and I see nothing to trigger the siren on your device picture.

closed 2024-06-12 by github-actions[bot]