dresden-elektronik/deconz-rest-plugin #7755: Hue outdoor sensor (SML004) reports different illuminance levels phoscon vs deconz?

jdeluyck, 2024-05-09:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I've recently gotten a new Hue Outdoor sensor, this one version SML004 vs an old SML002. 
The lux values are way off compared - I have them side by side, the old one reports 30000 in bright daylight (direct sun), the new one doesn't even go over 7000.

The values in Phoscon vs Deconz are also quite different (but I don't know if this is normal)
[image]
[image]

### Steps to reproduce the behavior

Get a new Philips Hue Outdoor sensor, model SML004, and use it.

### Expected behavior

Sensor values between two Hue's should be +- similar, not 20000 lux different, when next to eachother.

### Environment

- Host system: PC
- Running method: Home Assistent deCONZ Add-on
- Firmware version: 26660700
- deCONZ version (not Home assistant Addon version!): 2.25.3
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? Yes, a bluetooth USB stick.

ebaauw, 2024-05-09:
Phoscon shows the value in lux; the sensor's _Measured Value_ reports "light level", which uses a logarithmic scale.  From the Hue developer's documentation:
> Light level in 10000 log10 (lux) +1 measured by sensor. Logarithm scale used because the human eye adjusts to light levels and small changes at low lux levels are more noticeable than at high lux levels.

Different sensors, even of the same type, tend to report (slightly) different values; I've never found two sensors that would report exactly the same value when placed next to each other.  "With one watch a man knows what time it is, with two, he's never sure".

Anyway, this is not a bug in deCONZ nor in the REST API.  You might have a faulty sensor, but, in my experience, some variation in measured values comes with the territory.  Best practice is to calibrate each sensor by setting `config/tholddark` and `config/tholdoffset`, and only use `state/dark` and `state/daylight` in automations.

jdeluyck, 2024-05-09:
The problem being that I explicitly want to use these with the lux reading, as it's for steering sun blinds, so `state/dark` and `state/daylight` is too limited :(

Mimiix, 2024-05-09:
Closing as it's not a bug.

closed 2024-05-09 by Mimiix